stereolabs/zed-unity #121: unity version and ZED2

mattun, 2020-03-13:
2018.4.18f1 is good.
2019.3.xxxx is no good.

I hope that I can use this zed-unity on 2019.3.xxxx.

But I continue to use it on 2018.4.18f1.

Bvallon-sl, 2020-03-13:
Hi, Can you give me more details please?

Note that the Unity Plugin is only working with the standard rendering pipeline, not with LWRP/URP or HDRP.

mattun, 2020-03-13:
Hi
I am the beginner of Unity.

I checked again on 2019.3.5f1. When I click to play button, The Unity could recognize the ZED2 camera.  But when I build and run, I couldn't.
I remembered before I checked another in 2019.3.XXXX versions, "ZED_Rig_Mono" was ok, but "ZED_Rig_Stereo" was NG.

So I tried pre-version of Unity,2018.4.18f1 is good.

I always selected the 3D template, and not change the pipeline, use the default setting.

When to use ZED Plugin for Unity v3.0.1 appeared many messages, and restart.
But when to use ZED Plugin for Unity v3.1.0, appear the confirm for changing some values.

closed 2022-04-27 by github-actions[bot]