stereolabs/zed-unity #122: Zed PointCloud in Unity

DBrown12, 2020-03-17:
While using the pass through with the Zed_Rig_Stereo, we were able to use the Max Depth Range. I was wondering if there might be any information out there for using that Max Depth Range for the Point Cloud information. We tried adjusting in as is and it doesn't work. We also tried adjusting it in the shader and it didn't see to have any effect. Any suggestions would be wonderful!

ckorris, 2020-03-18:
You can do something similar to how it works in ZED_Forward_Lighting, where you call `discard` if the depth is beyond a certain range. You just have to go about it a bit differently.

Namely, the 3D position of the pixel (including depth) isn't available in the fragment shader as it stands; that's sampled in the vertex shader, but you'll want to call `discard` in the fragment shader. The only position that's currently passed to the fragment shader is the world position, _not_ the position relative to the camera, so you'll want to change that. 

Just tested this slightly modified version of the ZED_PointCloud shader and it worked: 

```
Shader "ZED/ZED PointCloud"
{
	Properties
	{
		_MainTex ("Texture", 2D) = "white" {}
		_Size("Size", Range(0.1,2)) = 0.1
		_MaxRange("Max Range", Range(1, 40)) = 40
	}
	SubShader
	{
		Pass
		{
			Cull Off
			CGPROGRAM
			#pragma target 5.0
			#pragma vertex vert
			#pragma fragment frag
			#include "UnityCG.cginc"
			
			struct PS_INPUT
			{
				float4 position : SV_POSITION;
				float4 color : COLOR;
				float3 normal : NORMAL;
				float depth : TEXCOORD0;
			};

			sampler2D _MainTex;
			float4 _MainTex_ST;
			
			sampler2D _XYZTex;
			sampler2D _ColorTex;
			float4 _XYZTex_TexelSize;
			float4x4 _Position;

			float _Size;

			PS_INPUT vert (appdata_full v, uint vertex_id : SV_VertexID, uint instance_id : SV_InstanceID)
			{
				PS_INPUT o;
				o.normal = v.normal;

				//Compute the UVS
				float2 uv = float2(
							clamp(fmod(instance_id, _XYZTex_TexelSize.z) * _XYZTex_TexelSize.x, _XYZTex_TexelSize.x, 1.0 - _XYZTex_TexelSize.x),
							clamp(((instance_id -fmod(instance_id, _XYZTex_TexelSize.z) * _XYZTex_TexelSize.x) / _XYZTex_TexelSize.z) * _XYZTex_TexelSize.y, _XYZTex_TexelSize.y, 1.0 - _XYZTex_TexelSize.y)
							);

				//Load the texture
				float4 XYZPos = float4(tex2Dlod(_XYZTex, float4(uv, 0.0, 0.0)).rgb ,1.0f);

				o.depth = XYZPos.z;

				//Set the World pos
				o.position = mul(mul(UNITY_MATRIX_VP, _Position ), XYZPos);

				o.color =  float4(tex2Dlod(_ColorTex, float4(uv, 0.0, 0.0)).bgr ,1.0f);

				return o;
			}

			float _MaxRange;
			
			fixed4 frag (PS_INPUT i) : SV_Target
			{
				if (i.depth > _MaxRange) discard;

				return i.color;
			}
			ENDCG
		}
	}
}
```

With that, you can change the maximum range by sliding the slider in the material properties. Out of the box, you can get to the material instance (which you can modify at runtime for immediate results) by double clicking on the Mat field in the ZEDPointCloudManager component. Or if you want to change it by code, you can use `mat.SetFloat("_MaxRange, [range you want])` from any script, where `mat` is the mat Material in ZEDPointCloudManager.

Duotun, 2020-11-05:
Works like a charm!

closed 2022-04-27 by github-actions[bot]