stereolabs/zed-unity #123: Getting OpenCV plugin to work with the ZedMini + Oculus

Duotun, 2020-03-21:
I've tried a few thing's to get OpenCV to run in the Oculus Rift with the Zed Mini. Finally I tried applying the OpenCV Script Dnn Object Detection Web Cam Texture Example, and Web Cam Texture To Mat Helper to the Frames under each eye in the Zed_Rig_Stereo. What happens though is I get error's informing me that the ZEDManager has already started the camera and because of this the OpenCV scripts error out. Is there a quick plug in play script for the Zed Mini to use OpenCV in Unity? If not any suggestion's for how I can fix the issue above?

ckorris, 2020-03-21:
The ZED Unity plugin has a sample for that. See the docs: https://www.stereolabs.com/docs/unity/using-opencv-with-unity/

The example scenes are built for ArUco marker detection, but the scripts for interfacing with OpenCV should be extendable to the rest of the plugin.

Duotun, 2020-03-23:
Thank you ckorris, this lead us to the ZEDToOpenCVRetriever.cs. We'll need to do some programming to get Yolo working with the script but this has definitely lead us in the right direction!

Duotun, 2020-03-23:
Keeping this open just in case we have any other questions!

33kiyu1, 2020-12-17:
I am also trying this method. Did you complete the script? Please let me know if you can.
I have ZED mini, Oculus, OpenCV for Unity.