stereolabs/zed-unity #124: Unity Object Detection failed to start. (Error: CORRUPTED_SDK_INSTALLATION )

LudwigVRtualX, 2020-03-23:
When starting the 2D or 3d Object Detection example scenes, I get the following error message:

```
Object Detection failed to start. (Error: CORRUPTED_SDK_INSTALLATION )
UnityEngine.Debug:Log(Object)
<startObjectDetection>d__296:MoveNext() (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:2628)
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
```

I tried this with the following combinations, all did not work:
- Unity 2019.3.5f1 + ZED SDK 3.1.1 + CUDA 10.2
- Unity 2019.3.5f1 + ZED SDK 3.1.1 + CUDA 10.0
- Unity 2019.2.12f1 + ZED SDK 3.1.1 + CUDA 10.2

All other example scenes work, the tools in the ZED SDK also all work. I always imported the ZED Plugin v3.1.0 in an empty scene.

I am on Windows 10, 64bit with a GTX 1070 and ZED 2.

Can anyone reproduce this isse or suggest a working setup?

Thanks.

Mitch-at-Beaudry, 2020-03-25:
Same issue with:
Unity 2019.3.5f1 + ZED SDK 3.1.0 + CUDA 10.2.89
Windows 10, 64bit + GTX 1080 +ZED 2

Was able to fix with Unity 2018.4.19f1 (LTS), but obviously would prefer to have a newer version of Unity working for this.

Canijo56, 2020-05-01:
Same here with ZED SDK 3.1.2 + CUDA 10.2.89
Windows 10, 64bit + 2x SLI GTX 1080 +ZED 2

Only working in 2018.4++, cant make it work in any 2019.

I tried also with ZED SDK 3.0 + Correspondant Unity Plugin and also with their Cuda 10.0 pairs. Cant make them work in 2019. (Uninstalling, rebooting, installing, rebooting every time).

obraun-sl, 2020-09-17:
One workaround for the Error: CORRUPTED_SDK_INSTALLATION inside Unity 2019.X :

Copy the dll of CUDA toolkit (located in <CUDA_PATH>/bin/) into the folder of the Unity Editor.
For example, if using the Unity Hub to install Unity editor version, it's located in : 
C:/Program Files/Unity/Hub/Editor/<Unity version>/Editor/ ,(for instance, <Unity version> =  2019.4.4f1

The reason is that the Unity plugin load a native C++ plugin (sl_unitywrapper.dll) that loads dynamically sl_ai.dll (ZED SDK) that is linked to CUDA library. It seems that starting from Unity2019, it cannot chained dependencies if they are not in the Unity folder.

lmbarns, 2020-09-24:
Could you name the exact DLL from CUDA to copy over? There are dozens of dll's in the bin folder and none of them resemble "cuda toolkit", is it cudart32_100.dll, cudart64_100.dll, cudann64_100.dll, etc, or all of them?

lmbarns, 2020-09-24:
Also I get the same error on Unity 2018.4 after trying a new project. Re-installed the SDK multiple times, 3.1.2 CUDA 10, I have cuda 9, 10, 11 on the machine, nvidia gtx 1070 gpu, latest Unity plugin. I have internet, Error: Corrupted_SDK_Installation

obraun-sl, 2020-09-25:
You need to copy the cudaXXX.dll (cudart64_xxx.dll)  + 
```
Also, you need to copy from ZED SDK/bin folder :
cuddn, myelin64n nvinfer,nvinfer_plugin, nvonnxparser.
```

krishnanpc, 2020-11-21:
@obraun-sl : Thanks for the solution. Note,  it takes a long while, the first time you load the OD module. But it works.

Kismetzc, 2021-07-20:
@obraun-sl 
The same question with
Windows10
Unity2020.3.14f1
Cuda11.0
SDK3.5.0
ZED2

I followed your instruction on my computer.
Copy from `cuda_path/bin` :
    `cudart32__110.dll,
      cudart64__110.dll
   `
from`ZED_SDK/bin`
```
   cudnn_adv_infer64_8.dll
   cudnn_cnn_infer64_8.dll
   cudnn_ops_infer64_8.dll
   cudnn64_8.dll
   myelin64_1.dll
   nvinfer.dkk
   nvinfer_plugin.dll
   nvonnxparser.dll
```

Then I copy them to  2020.3.14f1/Editor. It doesn't work.

obraun-sl, 2021-07-20:
@Kismetzc Can you provide the full path where you copied the dlls ?

Kismetzc, 2021-07-20:
> @Kismetzc Can you provide the full path where you copied the dlls ?

From CUDA_PATH: `C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin`
ZED_SDK: `C:\Program Files (x86)\ZED SDK\bin`
Unity: `D:\unity\2020.3.14f1\Editor`

And the unity hub is on`D:\unity\Unity Hub`

I named the folder named unity

Kismetzc, 2021-07-20:
By the way,
there are 
```
 object_accurate_2.1.model
 object_accurate_2.1.model_optimized-fbcaw-1-61-XXX
 object_accurate_2.1.model_optimized-fbcaw-1-61-XXX
```
in `C:\ProgramData\Stereolabs\resources`

obraun-sl, 2021-07-20:
Models are correct. You have the raw model (object_accurate_2.1.model) and the optimized model for your graphic card (object_accurate_2.1.model_optimized-xxxx)

The error is mostly coming from the fact that unity cannot load the sl_ai library. 
Is your unity path in the windows PATH ? (if you rename the folder, you need to update the PATH variable )

Kismetzc, 2021-07-21:
> Models are correct. You have the raw model (object_accurate_2.1.model) and the optimized model for your graphic card (object_accurate_2.1.model_optimized-xxxx)
>
> The error is mostly coming from the [...]

I updated my PATH and restarted my computer.
I add the PATH:
```
D:\unity\2020.3.14f1\Editor
D:\unity\2020.3.14f1
D:\unity
D:\unity\Unity Hub
```
In order to obtain the model,  I ran the cpp sample `Body Tracking`. Now I get the new model 
```
skeleton_accurate_2.0.model 
skeleton_accurate_2.0.model_optimized-aaaaw-1-61-11000-8200-7103
```
But the same error appeared again.  It can't detect the people.

I have tested the ZED2 camera on ubuntu18,  it works. So the ZED camera is no problem

Kismetzc, 2021-07-21:
I tried to reinstall the unity hub and unity2020.3.14f1 on `C:\Program Files`.
And  I followed your instruction again and copied the `sl_ai64.dll`,  `sl_zed64.dll` too.
Then I added 
```
C:\Progarm Files\2020.3.14f1\Editor  
C:\Progarm Files\2020.3.14f1
C:\Progarm Files\Unity Hub
```
to the PATH.

It didn't work. Does this problem occur only when installed through unity hub? Will this problem be avoided if unity is installed directly?

obraun-sl, 2021-07-21:
Can you confirm that it's working with the C++ samples ? (body tracking and object detection)

obraun-sl, 2021-07-21:
I always install Unity with the hub so the problem does not come from that part

Kismetzc, 2021-07-21:
> Can you confirm that it's working with the C++ samples ? (body tracking and object detection)

I did it again. And the samples didn't work. let me reinstall ZED SDK.

Kismetzc, 2021-07-21:
> Can you confirm that it's working with the C++ samples ? (body tracking and object detection)

## the output from build in VS2019
```
Build started...
1>------ Build started: Project: ZERO_CHECK, Configuration: Release x64 ------
1>Checking Build System
2>------ Build started: Project: ZED_Body_Tracking_Viewer, Configuration: Release x64 ------
2>Building Custom Rule C:/Program Files (x86)/ZED SDK/samples/body tracking/cpp/CMakeLists.txt
2>cl : command line warning D9002: ignoring unknown option '-std=c++14'
2>GLViewer.cpp
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(2507,1): warning C4819: The file contains a character that cannot be represented in the current code page (936). Save the file in Unicode format to prevent data loss
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(5723,16): warning C4251: 'sl::InitParameters::sdk_verbose_log_file': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(145): message : see declaration of 'sl::String'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(5775,19): warning C4251: 'sl::InitParameters::input': class 'sl::InputType' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(638): message : see declaration of 'sl::InputType'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(5790,16): warning C4251: 'sl::InitParameters::optional_settings_path': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(145): message : see declaration of 'sl::String'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(5799,16): warning C4251: 'sl::InitParameters::optional_opencv_calibration_file': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(145): message : see declaration of 'sl::String'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(5985,19): warning C4251: 'sl::PositionalTrackingParameters::initial_world_transform': class 'sl::Transform' needs to have dll-interface to be used by clients of struct 'sl::PositionalTrackingParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(2969): message : see declaration of 'sl::Transform'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6022,16): warning C4251: 'sl::PositionalTrackingParameters::area_file_path': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::PositionalTrackingParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(145): message : see declaration of 'sl::String'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6186,31): warning C4251: 'sl::SpatialMappingParameters::allowed_resolution': struct 'std::pair<float,float>' needs to have dll-interface to be used by clients of struct 'sl::SpatialMappingParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6079): message : see declaration of 'std::pair<float,float>'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6201,31): warning C4251: 'sl::SpatialMappingParameters::allowed_range': struct 'std::pair<float,float>' needs to have dll-interface to be used by clients of struct 'sl::SpatialMappingParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6079): message : see declaration of 'std::pair<float,float>'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6353,16): warning C4251: 'sl::RecordingParameters::video_filename': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::RecordingParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(145): message : see declaration of 'sl::String'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6529,35): warning C4251: 'sl::ObjectDetectionRuntimeParameters::object_class_filter': class 'std::vector<sl::OBJECT_CLASS,std::allocator<sl::OBJECT_CLASS>>' needs to have dll-interface to be used by clients of struct 'sl::ObjectDetectionRuntimeParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6529): message : see declaration of 'std::vector<sl::OBJECT_CLASS,std::allocator<sl::OBJECT_CLASS>>'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6535,39): warning C4251: 'sl::ObjectDetectionRuntimeParameters::object_class_detection_confidence_threshold': class 'std::map<sl::OBJECT_CLASS,float,std::less<sl::OBJECT_CLASS>,std::allocator<std::pair<const sl::OBJECT_CLASS,float>>>' needs to have dll-interface to be used by clients of struct 'sl::ObjectDetectionRuntimeParameters'
2>C:\Program Files (x86)\ZED SDK\include\sl/Camera.hpp(6535): message : see declaration of 'std::map<sl::OBJECT_CLASS,float,std::less<sl::OBJECT_CLASS>,std::allocator<std::pair<const sl::OBJECT_CLASS,float>>>'
2>C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp(135,45): warning C4244: 'argument': conversion from 'double' to 'int', possible loss of data
2>C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp(135,31): warning C4244: 'argument': conversion from 'double' to 'int', possible loss of data
2>C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp(184,25): warning C4244: 'argument': conversion from 'int' to 'float', possible loss of data
[... cut ...]
2>C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\main.cpp(189,20): warning C4996: 'sscanf': This function or variable may be unsafe. Consider using sscanf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
2>Generating Code...
2>ZED_Body_Tracking_Viewer.vcxproj -> C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\build\Release\ZED_Body_Tracking_Viewer.exe
2>Done building project "ZED_Body_Tracking_Viewer.vcxproj".
========== Build: 2 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
```

Kismetzc, 2021-07-21:
## THE ERROR LIST
```
Severity	Code	Description	Project	File	Line	Suppression State
Warning	D9002	ignoring unknown option '-std=c++14'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\build\cl	1	
Warning	C4819	The file contains a character that cannot be represented in the current code page (936). Save the file in Unicode format to prevent data loss	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	2507	
Warning	C4251	'sl::InitParameters::sdk_verbose_log_file': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	5723	
Warning	C4251	'sl::InitParameters::input': class 'sl::InputType' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	5775	
Warning	C4251	'sl::InitParameters::optional_settings_path': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	5790	
Warning	C4251	'sl::InitParameters::optional_opencv_calibration_file': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::InitParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	5799	
Warning	C4251	'sl::PositionalTrackingParameters::initial_world_transform': class 'sl::Transform' needs to have dll-interface to be used by clients of struct 'sl::PositionalTrackingParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	5985	
Warning	C4251	'sl::PositionalTrackingParameters::area_file_path': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::PositionalTrackingParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6022	
Warning	C4251	'sl::SpatialMappingParameters::allowed_resolution': struct 'std::pair<float,float>' needs to have dll-interface to be used by clients of struct 'sl::SpatialMappingParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6186	
Warning	C4251	'sl::SpatialMappingParameters::allowed_range': struct 'std::pair<float,float>' needs to have dll-interface to be used by clients of struct 'sl::SpatialMappingParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6201	
Warning	C4251	'sl::RecordingParameters::video_filename': class 'sl::String' needs to have dll-interface to be used by clients of struct 'sl::RecordingParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6353	
Warning	C4251	'sl::ObjectDetectionRuntimeParameters::object_class_filter': class 'std::vector<sl::OBJECT_CLASS,std::allocator<sl::OBJECT_CLASS>>' needs to have dll-interface to be used by clients of struct 'sl::ObjectDetectionRuntimeParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6529	
Warning	C4251	'sl::ObjectDetectionRuntimeParameters::object_class_detection_confidence_threshold': class 'std::map<sl::OBJECT_CLASS,float,std::less<sl::OBJECT_CLASS>,std::allocator<std::pair<const sl::OBJECT_CLASS,float>>>' needs to have dll-interface to be used by clients of struct 'sl::ObjectDetectionRuntimeParameters'	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp	6535	
Warning	C4244	'argument': conversion from 'double' to 'int', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	135	
Warning	C4244	'argument': conversion from 'double' to 'int', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	135	
Warning	C4244	'argument': conversion from 'int' to 'float', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	184	
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	508	
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	524	
Warning	C4244	'argument': conversion from 'double' to 'const T', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	680	
Warning	C4244	'argument': conversion from 'double' to 'const T', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	680	
Warning	C4244	'argument': conversion from 'double' to 'const T', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	680	
Warning	C4244	'argument': conversion from 'double' to 'const T', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	681	
Warning	C4244	'argument': conversion from 'double' to 'const T', possible loss of data	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\GLViewer.cpp	681	
[... cut ...]
Warning	C4996	'sscanf': This function or variable may be unsafe. Consider using sscanf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\main.cpp	184	
Warning	C4996	'sscanf': This function or variable may be unsafe. Consider using sscanf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	ZED_Body_Tracking_Viewer	C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\src\main.cpp	189	
```

Kismetzc, 2021-07-21:
## THE OUTPUT FORM DEBUG
```
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\samples\body tracking\cpp\build\Release\ZED_Body_Tracking_Viewer.exe'. Module was built without symbols.
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\bin\sl_zed64.dll'. Module was built without symbols.
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\setupapi.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\bcrypt.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\psapi.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin\cudart64_110.dll'. Module was built without symbols.
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\opengl32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\dependencies\freeglut_2.8\x64\freeglut.dll'. Module was built without symbols.
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\dependencies\opencv_3.1.0\x64\opencv_world310.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\comdlg32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\dependencies\glew-1.12.0\x64\glew32.dll'. Module was built without symbols.
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\nvcuda.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\urlmon.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\nvcuvid.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\vcomp140.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\concrt140.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\winhttp.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\IPHLPAPI.DLL'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\glu32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\winmm.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msvcr120.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_5.82.19041.1110_none_792d1c772443f647\comctl32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msvfw32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\avifil32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\avicap32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\iertutil.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\srvcli.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\netutils.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\msacm32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\winmmbase.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\wldp.dll'. 
'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. 
[... cut ...]
The thread 0x2a18 has exited with code 1 (0x1).
The thread 0x13b4 has exited with code 1 (0x1).
The thread 0x1898 has exited with code 1 (0x1).
The thread 0xa40 has exited with code 1 (0x1).
The thread 0x1c7c has exited with code 1 (0x1).
The thread 0x1bc0 has exited with code 1 (0x1).
The thread 0x29b4 has exited with code 1 (0x1).
The thread 0x1064 has exited with code 1 (0x1).
The thread 0x1f48 has exited with code 1 (0x1).
The program '[10952] ZED_Body_Tracking_Viewer.exe' has exited with code 1 (0x1).
```

Kismetzc, 2021-07-22:
> Models are correct. You have the raw model (object_accurate_2.1.model) and the optimized model for your graphic card (object_accurate_2.1.model_optimized-xxxx)
>
> The error is mostly coming from the [...]

In `Output from Debug`, there is `'ZED_Body_Tracking_Viewer.exe' (Win32): Loaded 'C:\Program Files (x86)\ZED SDK\bin\sl_ai64.dll'. Module was built without symbols.` It is true that the sl_ai library cannot be called

Kismetzc, 2021-07-22:
@obraun-sl Do you have any ideas to solve this problem？

obraun-sl, 2021-07-23:
You should launch the exe files (ZED_Body_Tracking_Viewer and object detection) directly with the exe file and not through visual studio. Furthermore, build in release mode, otherwise you might have std conflicts between debug and release (since zed sdk is built in release mode). 

Can you use DependencyWalker and open sl_ai.dll inside ? That will help to figure out what is missing

Kismetzc, 2021-07-23:
> You should launch the exe files (ZED_Body_Tracking_Viewer and object detection) directly with the exe file and not through visual studio. Furthermore, build in release mode, otherwise you might have [...]

I always build the samples in release mode. Using debug mode is to check out errors. 
I followed your instruction. But I got the same error. I will try to use dependcvwalker.

Kismetzc, 2021-07-26:
Using DependencyWalker, I found my PC missing some system DLLs. Then I fixed it and updated my win10. Finally, the sample(BodyTranking) can be launched on my PC！
Thanks for your help，@obraun-sl

closed 2022-04-27 by github-actions[bot]