stereolabs/zed-unity #126: Incomplete Mask for 2D Object Detection

Mitch-at-Beaudry, 2020-03-25:
The mask texture in the 2D Object Detection demo consistently does not include 90% of the arm of the detected person (see attached screenshot). The bounding box appears correctly; it scales to include the entirety of both arms.

Are there settings for the mask texture that I have missed, or is this an issue with the mask data itself?

[image]

Mitch-at-Beaudry, 2020-03-25:
Aha, that makes a lot of sense. Having that parameter would be very helpful for my use case.

Thanks for the fast response, especially right now. Hope the team is faring well~

Mitch-at-Beaudry, 2020-03-27:
What is the typical timeline for reviewing and releasing a new parameter? I'm assessing whether or not the object detection approach is appropriate for a product I'm developing, but time is a critical factor.

brontesprocessing, 2021-10-01:
@obraun-sl Could you please let me know whether the parameter has already been released in the SDK? If not - is it still planned?

closed 2022-04-27 by github-actions[bot]

tim-williams-bi, 2024-05-15:
Did this feature ever get added in? We are attempting to do something very similar for a green screen effect.

But the mask never covers the forearms and hands. We wouldn't mind trading a looser fit for it including arms and hands, if there was a parameter added as mentioned above.

Beyond a parameter from the SDK, we may just try taking the 2D joint poses below the elbow and creating a crude mask from that to overlay on the main mask.

SLJLMacarit, 2024-05-17:
Hi @tim-williams-bi 
Unfortunately, it's not been added, and I don't think it will be soon.

Your solution of creating a mask from the 2d joint poses is probably the way to go right now.

We're currently working on a way to ingest masks into the object detection, the same way you can ingest bounding boxes today. I don't have an ETA yet, but it will be included in a further release.

kapakrat, 2025-06-18:
Hello @SLJLMacarit , 
please do you have an update on this issue. Has the mask been fixed?
Thank you!

tim-williams-bi, 2025-06-18:
Using 2D joint poses to mask worked really well for us; I'd recommend that as the issue still exists in current ZED SDK

kapakrat, 2025-06-18:
Hello @tim-williams-bi , thanks a lot for your input!
Please could you explain the approach, share some code or papers how the 2D joint poses to mask works?
I need a full body 2D mask and I hoped this was included in SDK as shown in https://www.stereolabs.com/docs/object-detection

tim-williams-bi, 2025-06-19:
We took the 2D joint poses reported by the ZED in screen-space, then calculated capsule shapes around those poses. We then made a texture mask where those pixels were white, and the rest were black.

Combining this with the texture mask provided by the ZED can be used as an alpha; white pixels are the person, black is nothing.