stereolabs/zed-unity #129: ZED Mini tracking stopped working

JurjenBraam, 2020-05-06:
Hi there,

I just started to experiment with the ZED Mini when tracking stopped working in Unity 2018.4.19f1,
The status goes from TRACKING_OK to TRACKING_SEARCH multiple times per second.
Therefore, tracking just simply does not work.

I've tried the SDK version 3.1.2 and  3.0.3 both have similar results.

Camera firmware is 1523

Can't seem to figure out what the problem is.

JurjenBraam, 2020-05-07:
I suspect it's a IMU firmware problem, does anyone have a valid firmware I can try to flash?

obraun-sl, 2020-05-08:
Hi,
This is probably not a firmware issue but more a IMU configuration issue. 
You can try to reset the configuration by using ZED Calibration the following way: 
$ ZED\ Calibration.exe --cimu
and let the reset perform.

JurjenBraam, 2020-05-08:
This solved my problem!

closed 2020-05-08 by JurjenBraam