stereolabs/zed-unity #130: Add namespaces to ZED code

DanielPD, 2020-06-08:
Currently the ZED codebase doesn't seem to use namespaces everywhere. Some parts do (e.g. the SDK/NativeInterface classes use namespace "s1"), while the majority of classes don't (e.g. SDK/Helpers/Scripts). This may cause conflicts with other plugins and assets when imported in Unity.

Specifically, I recently ran into a problem using ZED and [UMA](https://assetstore.unity.com/packages/3d/characters/uma-2-unity-multipurpose-avatar-35611). UMA defines its own helper classes under its own namespace, which includes a class to set [ReadOnly](https://github.com/umasteeringgroup/UMA/blob/7fb84d4ebecc0f5e2d1d05a45b5867065ba4c7f6/UMAProject/Assets/UMA/Core/Extensions/DynamicCharacterSystem/UMAAssetBundleManager/GUI/ReadOnlyAttribute.cs) attributes. Similarly, ZED defines a [ReadOnlyAttribute](https://github.com/stereolabs/zed-unity/blob/50f42e09a6f4263f88ce949009a713f765a4abae/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Utilities/Utils.cs) class, **without** namespace, causing a conflict.

RomanCalderon, 2024-02-05:
Having a similar issue here. Namespacing all ZED scripts would be very helpful!

prof-thies-pfeiffer, 2024-11-11:
In particular the ReadOnly attribute (Util.cs) is used in many assets, e.g. Convai, and each time we encounter conflicts with ZED.

closed 2026-09-22 by Bvallon-sl