stereolabs/zed-unity #131: VR/AR passthrough - Tracking stability

DanielPD, 2020-06-15:
When using the ZED Mini with a VR headset I have the option to use ZED's inside-out motion tracking or the headset's built-in tracking solution. I have noticed some strange behaviour when comparing both of these options, and am wondering whether other people also encounter this and have found a better solution.

(NB. I am using an Oculus Quest connected through a USB link cable)

**ZED inside-out tracking**
Motion tracking is enabled in the ZEDManager. This is the default, as it works out of the box. This has the advantage of blending the virtual objects with the real world pretty convincingly, compensating for latency / timewarp, etc. to deliver a smooth image.

However, I experience significant drift over time, even when using spatial memory with a pre-loaded .area file. In practice, this means that within a single short run of my project virtual objects will not be where I left them. For example, just looking around the room will cause virtual objects to shift (up to a meter), often causing them to collide with real floors/walls/objects.

**Headset tracking**
Motion tracking is disabled in the ZEDManager. When a headset has its own tracking available, ZED uses this to position the virtual objects in the scene. My headset's tracking seems to be much more accurate and stable than ZED's own inside out tracking. I have not experienced any drift: virtual objects remain exactly where I place them in relation to the real world, even when walking and looking around during extended play sessions.

However, this also seems to disable ZED's smart compensation algorithms for dealing with image latency and timewarping. When looking around, the location of virtual objects are updated instantly in the frame, while the real-world camera feed lags behind several frames. This results in objects "wobbling" in place while the real-world camera images are "catching up" with the virtual objects. They seem completely disconnected with the real world. Additionally, the framerate of the camera seems quite a bit lower, resulting in a somewhat nauseating experience.

Ideally, I would like to use the headset's tracking, combined with ZED's intelligent merging algorithms to deal with the latency. Is this possible?

TateLampe, 2020-12-03:
I am having the same issues in a project I am working on. In my case, we are using external trackers but are still having the 'catching up' issues, where a user can circle their head and the 3d objects follow along albeit out of sync.

closed 2022-04-27 by github-actions[bot]

DanielPD, 2022-10-28:
Unfortunately, no we did not find a solution. Although I admit I stopped searching for one quite soon, and switched to Hololens for our project instead. Perhaps an interesting alternative could be the new Oculus Quest Pro headset which apparently also features integrated 3D RGB camera's for passthrough AR. Have not yet had a chance to try that feature, but the reviews seem promising!

FeldNico, 2022-10-28:
@DanielPD Thank you for your answer!
Unfortunately, one of my colleagues had the chance to try out the Oculus Quest Pro and stated that the image quality is worse than the ZED Mini and suggested keeping the ZED Mini. We now switched from the Oculus Quest 1 to a tethered VIVE Pro and Headset tracking, which made the stability a bit better. As we want the user to use both AR and VR simultaneously the Hololens 2 is not an option :/ 

For our next project, we hope for a Lynx-R1 or a Lenovo ThinkReality VRX.
Good luck with your projects!

DanielPD, 2022-10-28:
Ah unfortunate to hear the 3D RGB image quality of the Pro is underwhelming. Thanks, I'll keep an eye on the headsets you mentioned, might also be interesting for our lab!

Sounds like you're working on a cool project, good luck!