stereolabs/zed-unity #134: The ZED plugin cannot be loaded

mrbingo41, 2020-07-24:
Looks like the latest 3.2 SDK is not recognized by the latest Unity plugin.
The release archive website is down so there is no way to get 3.1 SDK.

jernejp21, 2020-07-24:
- Download ZED SDK 3.1 from [here](https://www.stereolabs.com/developers/release/3.1/)
- Download ZED Unity plugin 3.1 from [here](https://github.com/stereolabs/zed-unity/releases)

I did this last week and it worked for me. Before installing ZED SDK on Windows, remove all other versions.

mrbingo41, 2020-07-24:
Their server with older SDKs is down, this link doesn't work
https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/3.1/ZED_SDK_Windows10_cuda10.2_v3.1.2.exe

Can you share the exe file you downloaded somehow? Thanks.

jernejp21, 2020-07-24:
Try this link. This is for CUDA 10.2. 
[https://download.stereolabs.com/zedsdk/3.1/cu102/win](https://download.stereolabs.com/zedsdk/3.1/cu102/win)

This is for CUDA 10. 
[https://download.stereolabs.com/zedsdk/3.1/cu100/win](https://download.stereolabs.com/zedsdk/3.1/cu100/win)

mrbingo41, 2020-07-24:
It leads to the same broken url.

ckorris, 2020-07-25:
I just tried now (2 days later) and the links work for me. Can you try again?

mrbingo41, 2020-07-27:
Works now, thank you.

obraun-sl, 2020-07-28:
Note that the ZED SDK 3.2 is compatible with the staging_3.2 branch here : 
https://github.com/stereolabs/zed-unity/tree/staging_3.2

closed 2022-04-27 by github-actions[bot]