stereolabs/zed-unity #136: URP Build - ZED Mini

accounts1020, 2020-08-05:
Thank you for getting started with URP support, I realize it's not ready for prime-time, but I'm experiencing an issue that you might have come across:

95% of the time in Editor, the ZED initializes correctly, but every once in a while, the Game view will render a single pixel (stretched to the view). It appears to be sampled from the ZED passthrough camera, because that color shifts as i move the headset. 

Unfortunately when i make a PC build, that problem happens 100% of the time. 
No errors are showing up in the Player log. 

Any idea what could be causing this? I'm guessing maybe a race condition in the initialize sequence and rendering of the planes?

accounts1020, 2020-08-05:
Update: I've managed to isolate it to the Post Processing toggle on each eye's camera. 
Enabling Post Processing will break the build.

jernejp21, 2020-08-06:
I'm also using URP in my project and everything works for me. I can make build and works as expected. I have enabled post processing on both eyes. Which VR device are you using? I'm using Oculus Rift. To enable VR in Unity, I did:
- Project Settings -> Player -> XR Settings and enabled VR support (in 2019.4 this still works even if it's deprecated).

furmankl, 2021-02-11:
Hi, authors of the plugin, I also would like to thank for releasing the beta Version with URP support, I would like to know if there are any plans of releasing of a non-beta version anytime soon or if URP-support is maybe already included in the newest 3.4 version?

davidezhang, 2021-11-19:
Hi, when trying to get zed mini working in URP, I ran into a series of errors related to this: "RenderPipelineManager" does not contain a definition for "beginFrameRendering." I know this probably has to do with version conflicts but would really appreciate a pointer regarding this. Thanks! I'm using Unity 2018.4.8f1 and ZED Unity plugin 3.6.0.

closed 2022-04-27 by github-actions[bot]