stereolabs/zed-unity #138: Is the latest ZED SDK 3.2.2 not supported?

poorboy44, 2020-09-14:
I have the latest ZED SDK 3.2.2 installed, but when I go to import the Unity package (ZED_Unity_Plugin_v3.1.0.unitypackage) in Unity, I get the message. Is it necessary to downgrade my ZED SDK to 3.1 to use the Unity package?

```

The ZED plugin cannot be loaded. 
 Please check that you have ZED SDK v3.1 installed

 If the problem persists, please contact our support team at support@stereolabs.com

UnityEngine.Debug:LogError(Object)
sl.ZEDCamera:CheckPlugin() (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:799)
ZEDPluginInspector:UpdateWnd() (at Assets/ZED/Editor/Scripts/ZEDPluginInspector.cs:159)
UnityEditor.EditorApplication:Internal_CallUpdateFunctions() (at /home/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:310)
```

obraun-sl, 2020-09-14:
You can use this branch for 3.2 compatibility : 
https://github.com/stereolabs/zed-unity/tree/staging_3.2

arainbilal, 2020-10-05:
Hi @obraun-sl I have used the branch fro 3.2 compatibility with ZED SDK 3.2.2 with CUDA 11.0 (https://www.stereolabs.com/developers/release/). I can confirm that the SDK works fine. While using the simple example from the 3.2 branch, I find the error "DllNotFoundException: sl_unitywrapper"(see the attached screen shot). I can confirm that it is under the plugin/linux folder. My development platform is Ubuntu 18.04/Nvidia driver 455.23/Unity 2019.4.11f1 LTS. Just wondering if it is possible for you to please give me a hint or guidance or some tips to resolve this issue? Thanks. Happy to downgrade the SDK to resolve this issue, but I could not find link to download SDK 3.2.0 CUDA 11.0 on the website. 
[image]

obraun-sl, 2020-10-05:
Can you post the output of : 
$ ldd libsl_unitywrapper.so  when in the Plugins/linux folder ?

closed 2022-04-27 by github-actions[bot]