stereolabs/zed-unity #139: Object Detection failed to start. (Error: CORRUPTED_SDK_INSTALLATION ), even Unity 2018

lmbarns, 2020-09-24:
I've tried both SDKs 3.1 and 3.2, I have the Zed 2.

Tried the 3.2 branch that has been linked on here: https://github.com/stereolabs/zed-unity/tree/staging_3.2 after trying the 3.1.2 download from your main website. 

Every unity demo works except object detection.  

I tried copying over the CUDA files to the Unity editor folder as mentioned here https://github.com/stereolabs/zed-unity/issues/124  all that did was cause Unity to lock up as it initializes the sensor. 

Computer passes the zed diagnostics fine, gtx 1070, 3ghz cpu, sensor runs every other demo fine, can generate meshes, pin objects, etc but no matter what, the **object detection** does not work.

Unity versions I've tried 2018.3, 2018.4, 2019.2, 2019.4 none of them work with object tracking.

obraun-sl, 2020-09-25:
See my answer in thread #124

Bvallon-sl, 2020-09-30:
Hi, when you start the OD module for the first time, it downloads and optimizes the AI model, which can take few minutes. 
Try to run a cpp sample from the ZED SDK folder to have more informations about that.

Best,

dmapes123, 2020-10-09:
@lmbarns We're having the exact same issues. Did you find a solution? We've tried the full gamit of CUDA and Unity versions and it either just crashes Unity or gives the corrupted SDK using thread #124 Works great in SDK outside of Unity though.

closed 2022-04-27 by github-actions[bot]