stereolabs/zed-unity #142: Object detection example crashing

jdcart, 2020-10-12:
Hi, 

I'm using the staging 3.2 Unity plugin because I'm interested in using the Object Detection module inside Unity. I followed the instructions to get the plugin working by copying the DLLs from the CUDA and ZED folders into the Unity editor install folder. 

I don't get the "corrupted SDK" message anymore, however, when running any of the object detection samples, Unity crashes after a few seconds.

Here's the relevant part of my editor log:

Object Detection module started in 2 seconds.
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
<startObjectDetection>d__301:MoveNext() (at Assets\ZED\SDK\Helpers\Scripts\ZEDManager.cs:2655)
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
 
(Filename: Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs Line: 2655)

Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) sl.ZEDMat.dllz_mat_free (intptr,int) [0x0000a] in <ab981cd0065f4f669292f1d592ba1bbe>:0
  at sl.ZEDMat.Free (sl.ZEDMat/MEM) [0x00001] in C:\Perforce\ZED2_DynamicOcclusion\Assets\ZED\SDK\NativeInterface\ZEDMat.cs:375
  at ZEDManager.RetrieveObjectDetectionFrame () [0x00060] in C:\Perforce\ZED2_DynamicOcclusion\Assets\ZED\SDK\Helpers\Scripts\ZEDManager.cs:2745
  at ZEDManager.AcquireImages () [0x00153] in C:\Perforce\ZED2_DynamicOcclusion\Assets\ZED\SDK\Helpers\Scripts\ZEDManager.cs:2101
  at ZEDManager.ThreadedZEDGrab () [0x00095] in C:\Perforce\ZED2_DynamicOcclusion\Assets\ZED\SDK\Helpers\Scripts\ZEDManager.cs:2049
  at System.Threading.ThreadHelper.ThreadStart_Context (object) [0x0001f] in <9577ac7a62ef43179789031239ba8798>:0
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext,System.Threading.ContextCallback,object,bool) [0x00073] in <9577ac7a62ef43179789031239ba8798>:0
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext,System.Threading.ContextCallback,object,bool) [0x00004] in <9577ac7a62ef43179789031239ba8798>:0
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext,System.Threading.ContextCallback,object) [0x0002f] in <9577ac7a62ef43179789031239ba8798>:0
  at System.Threading.ThreadHelper.ThreadStart () [0x00014] in <9577ac7a62ef43179789031239ba8798>:0
  at (wrapper runtime-invoke) object.runtime_invoke_void__this__ (object,intptr,intptr,intptr) [0x0004f] in <9577ac7a62ef43179789031239ba8798>:0

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

0x00007FFDB766E738 (0x0000000000000000 0x0000000000000000 0x000000546ACAEEC0 0x0000000042C10101), ?size@SparseMat@cv@slutils@@QEBAHH@Z() + 0x7D8267 bytes(s)
0x00007FFDD9511F4E (0x0000000000000000 0x0000000000000000 0x000001C1C144C250 0x000000546ACAEEA0), dllz_mat_free() + 0xE bytes(s)
0x000001C1EF267E06 (0x0000000000000000 0x000001C1C144C250 0x000000546ACAEEA0 0x0000000000000000) <unknown module>
Crash!!!

bulshack, 2020-10-15:
Hello, I have the Zed 2 camera. I am trying to get the object detection to work as well on Unity. The object detection worked on the SDK version 3.1, with Cuda 10.2. We want the skeleton tracking feature from the SDK 3.2. Once the object detection sample runs unity stop responding. 

Currently using: 
Unity 2019.4.10f1
CUDA 11.0
ZED SDK 3.2.2
zed-unity-staging_3.2
Window 10 64bit

I ran SDK skeleton tracking sample so it calibrates the AI model before anything. I tried the DLL approach that worked on the 3.1, but not in the 3.2. We are just looking to get the head position using the skeleton tracking feature. We tried as well in different computers that have different GPUs. Currently, using GTX 1660 SUPER. We tried different cameras before, and our client really like the Demo of the skeleton tracking that Zed 2 offers. We think that this is the best skeleton tracking so far and we really want it to work for our project.

Bvallon-sl, 2020-10-16:
Hi, the crash seems to come from the 2D mask. I've created a new version of the Unity plugin without it that fix the issue.
A cleaner fix will be added with the 3.3 release but in the mean time you can use this version. 
You can find it here : https://github.com/stereolabs/zed-unity/tree/3.2_no_mask

If it does not solve the issue for you, do not hesitate to write me back.

Best

jdcart, 2020-10-20:
@Bvallon-sl Thanks, that worked fine for me.

closed 2020-10-23 by jdcart