stereolabs/zed-unity #143: Point cloud spatial mapping

JPlasteig, 2020-10-16:
Hi,
I was wondering how to achieve spatial mapping using point cloud in Unity. The SDK 3.1.0 is currently the lastest compatible with the unity plugin. Is this feature planned for later since this option is available using ZedFu?

I'm using:

- Unity 2019.3.1f1
- CUDA 10.2
- ZED SDK 3.2.2

Anyone got any info about that?
Thank you!

obraun-sl, 2020-10-16:
If you are referring to the fused point cloud mapping, it is available in the ZED SDK 3.2 with the unity plugin 3.2.
We provide a script ZEDFusedPointCloudManager.cs you can add in the project to visualize the current fused point cloud.

JPlasteig, 2020-10-20:
Thank you for your help ! I didn't notice the staging branch before since I was only looking for releases ...
However your solution is what I was looking for so I better close this issue !

closed 2020-10-20 by JPlasteig