stereolabs/zed-unity #144: ZED unity is not compatible with XR plugin management

daniel-halodi, 2020-10-16:
None of the VR functions work correctly due to being made for the XR device system in unity that obsolete and deprecated from unity 2020.1, and is replaced by the XR plugin management system, rendering it unusable.

DrFritzi, 2021-03-18:
Any news on this? The ZED Unity SDK is the only package blocking us from updating to the newer Unity versions.

Bvallon-sl, 2021-05-04:
Hi,

Sorry for the late reply.

You can take a look at the new branch called "[xr_compatibility_3.5]"(https://github.com/stereolabs/zed-unity/tree/xr_compatibility_3.5). It is now compatible with newer Unity versions and the new XR plugin management.
For the moment it keeps backward compatibility with the old system and we tried to duplicate the least amount of code possible  so it does not use that much of the new API (to avoid breaking everything for Unity2018 users).

We will wait for your feedbacks before merging into the master branch.

Best regards,
Benjamin V.

closed 2022-04-27 by github-actions[bot]