stereolabs/zed-unity #145: Skeleton 2D Keypoints

jdcart, 2020-10-22:
I've seen that in Unity currently you can only get the 3D keypoints from the tracked object. Is there any reason why the 2D keypoints are not exposed? Is there a plan to expose them in the next plugin release?

As a follow up question, is there any reason to use the 2D points over the 3D points? Is there maybe an accuracy, performance or detection distance improvement if using 2D over 3D or vice versa?

Thanks!

obraun-sl, 2020-10-23:
It was just to simplify the skeleton structure which already contains a lot of data and the 2D keypoints can be obtained from 3D keypoints by using the world to screen transform. 
But we can add it in next release.

Best is to use 3D points in Unity. Inside the ZED SDK, they are extracted from 2D images and then re-projected to the 3D world. 
They won't be any difference between 2D and 3D points (except that you loose the Z coodinates)

jdcart, 2020-10-23:
I thought as much, just wanted to make sure I wasn’t missing something. There is no need to expose the 2D points, as you said if needed, I can just use the same 3D points and project them to screen space.

Thanks for the reply!

closed 2020-10-23 by jdcart