stereolabs/zed-unity #149: Object Detection make CPU overload

nguyenanhtien2606, 2020-11-06:
Hi,
I need to use object detection for my project, I have updated to latest SDK (version 3.3.0). But all four detect mode are causing my CPU to overload (details in the image below).

My system info:
CPU: Intel Core I9-9900K
RAM: 16GB
GPU: GTX 1060 3GB

I just run default scene demo Object detection of ZED SDK. How I can fix it?

[image]
[image]
[image]

nguyenanhtien2606, 2020-11-12:
UPDATE!
I also test it in another strong hardware:
CPU: Ryzen 9 3900X
RAM: 32GB
GPU: GTX 1080 8GB

And I got the same result: 99-100% CPU for object detection.
Any solution for this?

Bvallon-sl, 2020-11-12:
Hi,

Sorry for the inconvenience. We are looking into it at the moment.

Best regards

Bvallon-sl, 2020-11-25:
Hi,

We've found the issue. It's coming from the version of opencv we used for the multithreading in OD/SK when the Object tracking option is enabled. It is not hardware related at all.

It will be fixed with the 3.4 release of the SDK.

The only workaround I can give you is to use the version 3.2.2 instead.

Best regards

nguyenanhtien2606, 2020-11-26:
> Hi,
>
> We've found the issue. It's coming from the version of opencv we used for the multithreading in OD/SK when the Object tracking option is enabled. It is not hardware related at all.
>
> It will be [...]

Hi,
Thanks for your answer. I'm using version 3.2.2, this version don't make overload when use "MULTI_CLASS_BOX" model. But "HUMAN_BODY" models still have this error. Hope it will be fixed soon.

Many thanks!

Bvallon-sl, 2020-12-01:
Hi,

We have release a 3.3.2 version of the ZED SDK that is fixing the CPU usage issue.
You can download it [here](https://www.stereolabs.com/developers/release/).

Best regards.

nguyenanhtien2606, 2020-12-03:
> Hi,
> 
> We have release a 3.3.2 version of the ZED SDK that is fixing the CPU usage issue.
> You can download it [here](https://www.stereolabs.com/developers/release/).
> 
> Best regards.

Thanks for new version! It works very well.

closed 2020-12-31 by nguyenanhtien2606