stereolabs/zed-unity #150: How do I get camera view projection matrix ?

AndreAhmed, 2020-11-11:
Hi,
How do I get camera view projection matrix in Unity ?

What I did so far
```
  zedManager.zedCamera.GetPosition(ref zedOrientation, ref zedPosition);

            Vector3 scale = new Vector3(1, 1, 1);

            Matrix4x4 m = Matrix4x4.TRS(zedPosition, zedOrientation, scale);
   sl.CameraParameters camparams = zedCam.CalibrationParametersRectified.leftCam;
        float fx = camparams.fx;
        float fy = camparams.fy;
        float cx = camparams.cx;
        float cy = camparams.cy;
        K.m00 = fx;
        K.m01 = 0;
        K.m02 = cx;
        K.m10 = 0;
        K.m11 = fy;
        K.m12 = cy;
        K.m20 = 0;
        K.m21 = 0;
        K.m22 = 1;

```

Bvallon-sl, 2020-11-12:
Hi,

The ZEDCamera class has a `Projection` attribute which stores the camera projection matrix. 

Best regards,
Benjamin Vallon

ckorris, 2020-11-19:
@AndreAhmed what Benjamin said will work and should be used, but if you do want to experiment with manually constructing the projection matrix, see how it's done with the ZED in [ZEDCamera.FillProjectionMatrix()](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCamera.cs#L1054-#L1082).

For more general understanding of Unity's projection matrix, you can find the layout in their docs [here](https://docs.unity3d.com/ScriptReference/Camera-projectionMatrix.html) (albeit with terrible naming conventions) and a very detailed user explanation [here](https://answers.unity.com/questions/1359718/what-do-the-values-in-the-matrix4x4-for-cameraproj.html).

closed 2022-04-27 by github-actions[bot]