stereolabs/zed-unity #156: Huge lag in skeleton tracking

Janne-CSE, 2021-01-12:
We're experiencing considerable lag when using the object detection module for skeleton tracking. There's about half second delay between the actual human movement and receiving the position data on OnObjectDetection. We are using skeleton tracking for human controller purposes in our products which means any too noticeably lag makes the sensor/sdk unqualified for us.

Is there any way to reduce the lag to make ZED viable option for skeleton tracking?

Bvallon-sl, 2021-01-15:
Hi,

The skeleton tracking might be very performance heavy and then cause some lag.

Did you experience this issue on a custom scene or while using the skeleton tracking example scene ?
Some parameters can impact the performance of the tracking.

The detection model, **_HUMAN_BODY_ACCURATE_** is more accurate but slower than the **FAST** one and requires a powerful GPU to be used.

I recommend disabling the body fitting if you do not need it as it slows the tracking as well. With the body fitting enabled, its computes every joints, even not detected ones. You will always have a full skeleton.

The resolution/depth mode also impacts the performance.

Of course, theses parameters will also impact the accuracy of the tracking while speeding it.

You can also try to directly run the cpp sample to see if additional lags come from the Unity scene.

Best regards,
Benjamin Vallon

Janne-CSE, 2021-01-18:
Hi, thank you for the response.

The problem exists in both example scene and custom scene. I already use the HUMAN_BODY_FAST model and it seems the other settings you mentioned were set to gain high performance as well. I think the delay in the cpp sample was lower by 100 ms at very most. That being said, there's still like 300 ms too much delay for our needs.

This in unfortunate, but I'll keep my eyes on ZED's updates. There's lots of potential in it for different purposes and the support is way more responsive than with other sdk's/sensors. :)

closed 2022-04-27 by github-actions[bot]