stereolabs/zed-unity #157: How to get full viewport with AR pass-through

dorcatty, 2021-01-14:
Dear all, 

I am testing Zed M camera with Unity3D and Oculus Quest with Oculus Quest Integration. 
I can have the dual view from Zed Camera and AR content via Oculus Link Cable.
However, the viewport has some 'Black cropped' area, such as:

https://gfycat.com/deafeningornerykawala

do we have some solution to view AR Pass-through without black area ? 

thx

obraun-sl, 2021-01-15:
Hi,
It's not really possible since the FoV of the ZED-M and the Fov of the Quest are not the same (Quest is bigger). 
There is potentially a hack by changing the distance of the rendering plane, but that's not really recommended since images will look distorded.
You can change the parameter here : 
https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/MR/ZEDMixedRealityPlugin.cs#L431
Increase the value (default = 1.0f) or even increase the scaleFromZED value.

dorcatty, 2021-01-15:
thanks a lot your quick response.
do we have any camera like Zed Mini gives a larger FOV ?
or have you tested with other VR headset ?which goggle has the similar FOV
as Zed Mini ?

Thanks a lot

crud89, 2021-05-26:
To a certain extent, you can influence the FoV within the HMD by setting [`XRDevice.fovZoomFactor`](https://docs.unity3d.com/ScriptReference/XR.XRDevice-fovZoomFactor.html). However, similar to "stretching" the view plane to match the vision, this also changes the proportions of the user vision, so objects appear larger than they actually are.

closed 2022-04-27 by github-actions[bot]