stereolabs/zed-unity #158: Missing Entry Point in sl_unitywrapper

falichs, 2021-01-26:
Hi there,
i'd love to use the region-based automatic Gain/Exposure Mode on the ZED Mini. 

Unfortunately it seems to me that `[DllImport(nameDll, EntryPoint = "dllz_get_roi_for_aec_agc")]` from **ZEDCamera.cs** points to nowhere.
I'm using [DLL Export Viewer](https://www.nirsoft.net/utils/dll_export_viewer.html) to inspect the **sl_unitywrapper.dll** and it seems like the entry point is indeed missing.

Will this feature be added soon? 
It seems like the main ZED SDK [already supports](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#aa266794188e5b07018a6252fb5f5d78e) it.

Bvallon-sl, 2021-01-26:
Hi,

Thanks for reaching us out.
Indeed the function was missing in the dll.
It will be added in the 3.4 release of the plugin, which will be available soon.

Best regards,
BV

Bvallon-sl, 2021-01-27:
Hi again,

The 3.4 release is now available (https://github.com/stereolabs/zed-unity).
Can you confirm that the feature is now working correctly ?

Thanks for you help.

Best regards,
BV

falichs, 2021-01-29:
Thank you so much for the quick response!

I'm still in the process of upgrading my project to v3.4

once i'm done i will let you know if it works

falichs, 2021-02-11:
Thanks! It works!

closed 2021-02-11 by falichs