stereolabs/zed-unity #162: Sphere is being shaked ,when I add sphere in scene with ZED_Rig_Stereo, using HMD.

YuyaMurai, 2021-02-03:
I'm getting started with "https://github.com/stereolabs/zed-unity".
And I stucked with the chapter "Creating Your First Project".

・Right-click the Hierarchy and click 3D Object -> Sphere to put a basic sphere into the scene. Position it about a meter in front of the ZED camera rig.
・Select the Directional Light in your scene. Add a ZED Light component.
・Set the Directional Light to point straight down by setting its X Rotation to 90 in its Transform component.
・Run the scene. You'll see the sphere floating in the real world, and casting a shadown below it.

As a result, the sphere is being shaked ,and Icannot see the sphere as 3D object.
How can I fix it?

YuyaMurai, 2021-02-03:
[image]

obraun-sl, 2021-02-03:
Make sure the camera is physically attached to the HMD, so that they have the same movement. 
You can also try to uncheck "set IMU Prior in AR" in the advanced settings of the ZEDManager/ZED_Rig_Stereo inspector to see if it's better.

YuyaMurai, 2021-02-03:
I checked both of your advises but do not working.
However , I found the reason why the sphere is shaking and changing two places every frame.
The reason is that X,Y,Z-axes are converting every frame, when I use "ZED_Rig_mono" and "ZED_Rig_stereo".

obraun-sl, 2021-02-03:
There is maybe an issue with the IMU of your camera (Is your camera a ZED-M ?)
--> Try to check with the SDK tools ZED Sensor Viewer to see if the movement of your camera on the screen is coherent with the physical movement. 
If not, contact the support so that they can help you reset the IMU of the camera. (support@stereolabs.com)

YuyaMurai, 2021-02-03:
Yes, my camera is ZED-Mini.
I checked ZED Sensor Viewer and the sensors data were good, however camera is always switching left and right side per every frame.
[image]
[image]

obraun-sl, 2021-02-03:
Something might be wrong with the IMU calibration.
Try to execute the software attached in the zip folder. Don't move the camera while the software is running.
[ZED_M_IMU_KlibReset.zip](https://github.com/stereolabs/zed-unity/files/5917814/ZED_M_IMU_KlibReset.zip)

YuyaMurai, 2021-02-04:
Thank you!
I'll check it now.

YuyaMurai, 2021-02-04:
And it worked perfectly!
Thank you very much!

closed 2021-02-04 by YuyaMurai