stereolabs/zed-unity #165: Spatial mapping mesh texture not working in Unity 2020.1.8f1

jamestkpoon, 2021-02-12:
Running the default rendering pipeline, Ubuntu 18.04. With the "Texturing" option set to true, resultant texture .png is just a large empty image. Actual mesh geometry looks fine.

Sorry if it's something obvious, but is it something simple like changing the rendering pipeline, etc?

jamestkpoon, 2021-02-15:
Additional information: using SDK 3.4.0. Mapping via Python API generates textures correctly, and Unity imports them without issue.

Bvallon-sl, 2021-02-16:
Hi,

Are you on Linux ? If so, there was an issue in the wrapper that I just fixed. If you pull the master branch, the issue should be fixed now !

Best regards,
Benjamin Vallon

jamestkpoon, 2021-02-16:
Hi, 

First of all, thank you very much for the fast response!

Yes, I'm on Linux. Replaced libsl_unitywrapper.so as it seems to be the only file that has changed, then I restarted Unity. Unfortunately there is now an error:

The ZED plugin cannot be loaded. 
 Please check that you have ZED SDK v3.4 installed

ldd output shows library links are all set. Putting  the old .so back in and restarting Unity does not result in this error

Bvallon-sl, 2021-02-16:
Hi,

You are right, the lib was build on Ubuntu20, and then was not working on previous versions.

You are now pull the master branch or replace your lib by this one.

Best regards,
Benjamin Vallon

[libsl_unitywrapper.zip](https://github.com/stereolabs/zed-unity/files/5990057/libsl_unitywrapper.zip)

jamestkpoon, 2021-02-17:
That fixed it. Thank you!

closed 2021-02-17 by jamestkpoon