stereolabs/zed-unity #169: Occlusion for real-world frame

Duotun, 2021-03-06:
Sorry to bother you a little bit. I am wondering how could I setup the zedmini in Unity that the real-world frame will be always on the top of virtual objects? It seems that even I click on the depth occlusion, some virtual objects will still be in the front of real-world frame. 

Depth Occlusion: When enabled, virtual pixels can be covered up by real pixels, allowing a virtual cube to be behind your real table, for example. Turn off to make virtual objects always appear over the real world.

I haven't have a careful try but I may think the rendering queue may could solve this?

Thank you!

Bvallon-sl, 2021-03-08:
Hi,

Thanks for reaching us out.

What you can do is modifying the "no depth occlusion" option of our plugin to do exactly what you want.
The depth occlusion sets the depth of the frame accordingly ([here](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader#L115). You can just invert the result by modifying that value.

However, I don't understand what you are trying to achieve. By doing this, virtual objects won't be visible at all. So, the easiest solution would be to just not render them.

Best regards,
BV

Duotun, 2021-03-08:
Thx for your reply! What I am trying to achieve is that when rendering the real-world frame partially like this, I would like it to be the most front and others will be totally rendered by virtual objects. This following image will illustrate my problem better: 
[image]

closed 2022-04-27 by github-actions[bot]