stereolabs/zed-unity #172: Plugin cannot find ZED SDK

HainingLuo, 2021-03-24:
Hi all,

I am new to zed-unity integration and here is what I got after importing the plugin and hitting the play button in Unity;

```
DllNotFoundException: sl_unitywrapper
sl.ZEDCamera.CreateCamera (System.Int32 cameraID, System.Boolean verbose) (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:870)
ZEDManager.Awake () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:1784)
```
```
The ZED plugin cannot be loaded. 
 Please check that you have ZED SDK v3.4 installed

 If the problem persists, please contact our support team at support@stereolabs.com

UnityEngine.Debug:LogError (object)
sl.ZEDCamera:CheckPlugin () (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:823)
GUIMessage:Awake () (at Assets/ZED/SDK/Helpers/Scripts/Utilities/GUIMessage.cs:122)
```
```
NullReferenceException: Object reference not set to an instance of an object
ZEDRenderingPlane.OnRenderImage (UnityEngine.RenderTexture source, UnityEngine.RenderTexture destination) (at Assets/ZED/SDK/Helpers/Scripts/Display/ZEDRenderingPlane.cs:1327)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr) (at /home/bokken/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:197)
```

This is on Ubuntu 18.04 with ZED Plugin for Unity v3.4.0, ZED SDK 3.4.2, CUDA 11.0 and Unity3D 2019.4.22f1. ZED Diagnostic shows everything is fine. 

Anyone had same issue before? Any suggestion is appreciated! Many thanks in advance.

Bvallon-sl, 2021-03-26:
Hi,

This issue should have been fixed in a recent commit. Download the plugin directly on the master branch and it should be working fine.

Best regards,
Benjamin V

HainingLuo, 2021-03-27:
> Hi,
> 
> This issue should have been fixed in a recent commit. Download the plugin directly on the master branch and it should be working fine.
> 
> Best regards,
> Benjamin V

Hi Benjamin,

Thanks for your reply. I tried to clone the repo directly from master branch and opened the unity project within it. The errors are exactly the same as when I download the plugin. Is it possible that I need to install some other dependencies or add zed sdk path to some configuration files? 

Many thanks!

obraun-sl, 2021-03-27:
Hi,
Can you type in a terminal (in the folder where libsl_unitywrapper.so is located) : 
$ ldd libsl_unitwrapper.so 
and post the output here.

HainingLuo, 2021-03-27:
> Hi,
> Can you type in a terminal (in the folder where libsl_unitywrapper.so is located) :
> $ ldd libsl_unitwrapper.so
> and post the output here.

Hi, 
Here is the output:
```
	linux-vdso.so.1 (0x00007ffffc7d2000)
	libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007f2906398000)
	libcuda.so.1 => /usr/lib/x86_64-linux-gnu/libcuda.so.1 (0x00007f2904ee5000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2904cc6000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2904ac2000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f29048ba000)
	libusb-0.1.so.4 => not found
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2904531000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2904193000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2903f7b000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2903b8a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f290e7e8000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f2903958000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f290373b000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f29034d3000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007f2903263000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f290304b000)
	libnvcuvid.so.1 => /usr/lib/x86_64-linux-gnu/libnvcuvid.so.1 (0x00007f2902ac5000)
```
I noticed that libusb was not found. Does it mean that the camera was not detected? I tried to swap to other usb sockets but didn't have any luck.
Thanks!

obraun-sl, 2021-03-28:
Yes, the problem come from libusb-0.1.so.4. Since it's not found on your PC, the load of sl_unitywrapper is unsuccessful --> error.
To solve this, simply call : 
$ sudo apt install libusb-dev
This should install the missing dependencies.

HainingLuo, 2021-03-29:
Thanks Obraun! I installed libusb and it is working now!

closed 2021-03-29 by HainingLuo