stereolabs/zed-unity #174: Listening to stream on Unity in Windows results in crashes

jamestkpoon, 2021-04-13:
About the system:
- Windows 10
- Cuda 11.1, Zed SDK 3.4.2
- Unity 2020.3.2f1 URP
- latest Zed Unity package
- Zed 2

I would like to render point clouds for VR. When the Zed2 is plugged into the PC, everything works great. When listening to a stream from another Zed2 however, Unity crashes with no explanation after a few seconds to a minute. What I am doing with the Zed SDK doesn't seem to affect this; even just running the Manager with no image or point cloud renderers active still causes crashes.

Any suggestions are welcome. Thanks!

Bvallon-sl, 2021-04-14:
Hi,

First, I would recommend to use Unity2019 as the current Unity plugin is not compatible with the new Unity input system introduced in 2020 (we plan to change that in the future).

Then, regarding your crash, can you send me your crash report ? it's available at
`C:\Users\USERNAME\AppData\Local\Temp\Unity\Editor\Crashes`

Best regards,
Benjamin Vallon

jamestkpoon, 2021-04-14:
Hi again, Benjamin! Appreciate the help greatly.

I will try go back to Unity 2019, and hope it doesn't break anything else going on in that project.
In the meantime, unfortunately there doesn't seem to be a crash folder generated for these crashes. The ones I have seem to be from the Oculus plugin on application exit. Nevertheless I have attached a recent one. I also ran the ZedManager in an empty Scene and attached the Editor log in case it helps.

[Crash_2021-04-13_110840137.zip](https://github.com/stereolabs/zed-unity/files/6309915/Crash_2021-04-13_110840137.zip)
[Editor.log](https://github.com/stereolabs/zed-unity/files/6309916/Editor.log)

jamestkpoon, 2021-04-16:
This issue seems to only exist over our wifi, as plugging the Zed2 into another computer on the same wired switch does not reproduce these crashes and everything runs as expected.
It could be corrupted incoming data causes unexpected behaviours such as write access violations in the decoder? We experienced such a behaviour when testing turbojpeg decompression some time ago.

jamestkpoon, 2021-04-20:
Update: Rolling the SDK on the receiving computer back to 3.4.1 seems to avoid crashes but introduces a point cloud rendering delay of >5 seconds

Bvallon-sl, 2021-04-26:
Hi,

I'm sorry for the lack of updates.

The 35 release improved the streaming stabitility on Windows.
Can you try it out and come back to me if it does not solve your issue?

Best regards,
Benjamin V

jamestkpoon, 2021-04-26:
We were able to get a stable setup by rolling back the SDK on the receiving PC to 3.4.1 .
When I have time to try out 3.5, I'll get back to you here. Thanks for the update!

closed 2022-04-27 by github-actions[bot]