stereolabs/zed-unity #177: ZED disconnect logic not running

jdcart, 2021-04-30:
When the camera is disconnected the error reported in ZEDManager.AcquireImages() is "INVALID_FUNCTION_CALL" and there is no code that handles that case. "CAMERA_NOT_DETECTED" is never reported so isDisconnected is never set to true, and in turn, none of the reconnection logic is fired.

My workaround was to check for "CAMERA_NOT_DETECTED" || "INVALID_FUNCTION_CALL" and the camera reconnections work properly, however, not sure if this will cause other issues.

Please advise if this is an acceptable "workaround", thanks!

Bvallon-sl, 2021-05-03:
Hi,

Thanks for reaching us out.

Yes, this is totally acceptable but note that the grab() function returns "INVALID_FUNCTION_CALL" if the camera is not opened.
You can call the ZEDManager.Reset() function when the camera is reconnected to reopen the camera.

Best regards,
Benjamin Vallon

jdcart, 2021-05-03:
Benjamin,

I should mention that this is using the out-of-the-box ZED manager, so it should be something that needs to be addressed in the official release of the Unity package.

Thanks for confirmation that this is ok.

Regards,

Juan

closed 2021-05-03 by jdcart