stereolabs/zed-unity #178: When using AR, the Translation value works in reverse.

hagsac, 2021-05-06:
When using AR, the Translation value works in reverse.
We ask again as the question is unanswered.

The rotation value is normal, but if you move the Zed2 camera to the horizontal or vertical axis, the AR target will follow. It seems that the value of Translation is inversely transformed and operated.
When manufacturing ZED, is it possible to produce defective products in this category?

There was no answer a few weeks ago, so I'm asking again. Please help me.

Bvallon-sl, 2021-05-06:
Hi,

Do you mean that the virtual objects are moving with the camera and not staying still in the world ?
If so, make sure you enabled the camera tracking in the ZED Manager inspector window.

Best regards,
Benjamin  V.

hagsac, 2021-05-06:
Camera tracking and all settings are OK.
This issue also occurs in unity and basic zed samples.
The signal of the internal position sensor of ZED2 that I have seems to be connected in reverse.

This is normal if you create a virtual object and change the camera angle only in place.

However, if you move the camera itself, the virtual object follows in the same direction that moves slightly.
It feels like Translation value (X,Y,Z)is inverse.

closed 2021-05-06 by hagsac