stereolabs/zed-unity #180: Unity crashes while AI model is generated!

bastidererste, 2021-05-10:
[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes   |/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-\|/-Stacktrace:

crash!

#124 helped to get unity up and running until tis point.
Unity 2020.3.7f1
SDK 3.5
ZED2

bastidererste, 2021-05-10:
thx for lookin in to this!

the binary ZED_Object_detection_birds_eye_viewer.exe doesn't run either... heres the output:

C:\Program Files (x86)\ZED SDK\samples\bin>ZED_Object_detection_birds_eye_viewer.exe
[ZED][Init] Depth mode: ULTRA
[ZED][Init] Video mode: HD1080@30
[Sample] Object Detection: Loading Module...

[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes  -
C:\Program Files (x86)\ZED SDK\samples\bin>

bastidererste, 2021-05-10:
i also ran the python version, and it works

Bvallon-sl, 2021-05-10:
Hi,

Can you try different Detection model (Multi_class_box, Multi_class_box_accurate ..) to see if they are all causing a crash ?

Thanks for your feedbacks.

Best regards,
Benjamin Vallon

bastidererste, 2021-05-10:
they are all crashing in unity

bastidererste, 2021-05-10:
did delete the folder contents.

it downloads the models.

but then:

C:\Program Files (x86)\ZED SDK\samples\bin>ZED_Body_Tracking_Viewer.exe

[ZED][Object Detection] AI model not found, downloading  \Done
 Done
[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes   Done[ZED] [Object Detection] MODEL_NOT_FOUND
[Sample][Error] enable Object Detection  | CORRUPTED SDK INSTALLATION : Corrupted SDK detected. Please reinstall the ZED SDK.
Exit program.

Bvallon-sl, 2021-05-11:
Hi, 

We had some server issue yesterday, so the model you downloaded might be corrupted.
I joined the Multi_Class_Box_Fast model to this message.
Can you clean the ressources folder again, add the model I joined, and try to run a sample with the Multi_Class_Box_Fast model ?

Best regards,
Benjamin Vallon

[objects_performance_2.1.zip](https://github.com/stereolabs/zed-unity/files/6457769/objects_performance_2.1.zip)

bastidererste, 2021-05-11:
cant find Multi_Class_Box_Fast :-/
there is a Multi_Class_Box though

and unity still crashes

bastidererste, 2021-05-11:
when i copy the model to resources, set the model to Multi_Class_Box ZED_Object_detection_birds_eye_viewer goes some lines further, but still crashes

`101      ObjectDetectionRuntimeParameters detection_parameters_rt(detection_confidence);
`

bastidererste, 2021-05-11:
> unity plugin : >>>>https://github.com/stereolabs/zed-unity/releases

thats's what i use

> Unity 2021.1.6f1
that's what im trying right now..

bastidererste, 2021-05-12:
nope... didnt work

object detection doesnt work in unity and the ZED_SDK Examples.

I can run any other example but body tracking and object tracking...

bastidererste, 2021-05-12:
the cpp example of object_detection_birds_view doesnt run

the python example of object_detection_birds_view runs without any issue 

thats weird

Bvallon-sl, 2021-05-17:
Hi,

Sorry for the delay,

Can you verify both cpp and python sample you are running are using the same detection model ?
If not,  can you tell which detection models are working ? It would be nice to test it using a cpp sample.

Thanks for your help.

Best regards,
Benjamin Vallon

bastidererste, 2021-05-17:
both versions of object_detection_birds_eye_view (cpp, py) use DETECTION_MODEL.MULTI_CLASS_BOX

cpp crashes
py runs

closed 2022-04-27 by github-actions[bot]