stereolabs/zed-unity #181: Problem with AR object slip phenomenon.

hagsac, 2021-05-13:
Thank you always for your kind reply.
Referring to the ZED2 tutorial, create a simple object.
There is a problem when the camera is moved.

[image]

A slight slip occurs about the motion of the object on the camera.
This slip phenomenon is a phenomenon that moves along the vertical or horizontal direction of the camera.

**However, after turning off Visual-inertial Tracking in Advanced Setting in Camera_ring, it operated normally without any slip phenomenon of the object.**
No documents were found for that advanced setting.
If the above setting is enabled, please tell me how it affects the execution of the tutorial above.

[image]

Bvallon-sl, 2021-05-17:
Hi,

Thanks for reaching us out.

This **Visual-inertial Tracking** parameter is the equivalent of the **PositionalTrackingParameters::enable_imu_fusion** parameter of the ZED SDK Api.

Your IMU calibration might be incorrect which can explain the really bad result you are getting when enabling it.

You can try to use the ZED Calibration tool (available in the ZED SDK/tools/) to reset the IMU and see if it solves your issue.

Best regards,
Benjamin Vallon

hagsac, 2021-05-17:
I've found many ways, but I have the following problems.
I feel uncomfortable to implement the part about the calibration due to the following matters.

[image]
Because my PC monitor is curved monitor. Is ZED calibration influenced by curved monitors?
Also, is IMU calibrated when using the camera calibration tool?
First of all, I want to change it to a perfect default, but Tool doesn't seem to have that function.

So I tried the following methods.
ZED Explorer attempted to update IMU or Camera Firmware.
[image]

However, it was not updated normally.
[image]

After a long time on "2/7,"
[image]

In the event of an update failure, it is represented as '-1', which means that the ZED Driver must be released and reconnected from the window control panel to return to normal.

[image]

Camera update attempt is also a failure.
[image]

[image]

Bvallon-sl, 2021-05-17:
Hi,

If you run the ZED Calibration tool with the **--cimu** argument in a console window, it will only run the IMU calibration part and skip the rest. You do not have to worry about it.

Best regards,
Benjamin Vallon

hagsac, 2021-05-17:
Is this how everything works? I don't think anything has happened.
I'm sorry to bother you with so many questions.
[image]

Bvallon-sl, 2021-05-17:
Hi,

Are you using an old version of the ZED SDK ?
Can you try it with 3.5 ?

Best regards,
Benjamin Vallon

hagsac, 2021-05-17:
We are using cuda 11.1 and SDK 3.5 cuda 11.1.
[image]

However, the results are as follows.
But all the examples are working fine.

And cuda 11.1 is installed normally.
I reinstalled SDK 3.4 cuda 11.1 version and checked it.
The image below is the evidence.
[image]

Of course, cmd IMU calibration is the same symptom in both versions 3.4 and 3.5.. ...............

Symptoms of the above cuda installation problem are SDK 3.5 cuda 11.1 and
SDJ3.5 cuda10...I understand that the same problem occurs in .
It seems to be a common problem in 3.5.
I checked in both places because I am using two PCs.

hagsac, 2021-05-17:
Among the above questions, IMU or CAMERA's
Isn't there a problem with the firmware not being updated?

hagsac, 2021-05-17:
cuda ver. check

[image]

hagsac, 2021-06-03:
Good morning ~!
Please replay to my comments.

It seems that the existing AR problems
And SDK 3.5 problem  also need to be resolved.

I think the sdk 3.5 problem  is a problem  that all users have.

1.  Ar problem
2.  Sdk 3.5 problem
3.  Zed2 update problem

P-yver, 2021-06-03:
Hi,
Indeed the IMU firmware can not be updated from Windows for the moment (only working under Linux) but your issue does not come from it (either 776 or 777).
We also faced the IMU drift and are currently working on a patch to fix it, I will let you know when it is online.

hagsac, 2021-06-03:
Thank you for anshwer.

Your answer made me feel more at ease. :)

Please also answer the verification problem  when installing sdk 3.5.
After installing sdk 3.5, we did not find any major problems in the
examples except for AR problems, but it is not normal in the verification.
Are sdk 3.5 installation issues not having any impact when using zed2 ?

closed 2022-04-27 by github-actions[bot]