stereolabs/zed-unity #182: ZED mini + VIVE setup

AndreiaValente, 2021-05-18:
I just got a ZED mini to do some AR passthrough with the original HTC VIVE.

I followed the tutorial and just dragged the ZED_Rig_Stereo prefab to the scene.
When I run the project, the Body element just drifts like it was one screen. 
A video example of the problem: [https://www.youtube.com/watch?v=t_JQsA2z1CA](url)

I'm using all the default setting on the ZED manager. 

On the OpenVR tab I'm using:
Application Type=Scene 
Stereo Rendering Mode=Multi Pass
Mirror View Mode=OpenVR

I'm not sure what I'm doing wrong on the setup. Any help would be appreciated.

Unity 2020.2.1f1
SDK 3.5

Bvallon-sl, 2021-05-18:
Hi,

The master branch has some compatibility issues with the XR Plugin management system. Can you try to use [this branch](https://github.com/stereolabs/zed-unity/tree/xr_compatibility_3.5) instead ? It should be working correctly.

Best regards,
Benjamin Vallon

crud89, 2021-05-20:
I am experiencing the same issue and tried the changes at the xr compatibility branch. However, it did not resolve the problem. Interestingly, this only occurs within the HMD, not in the game preview in Unity Editor. 

I am on Unity 2021.1.7f1, ZED SDK 3.5.

Bvallon-sl, 2021-05-20:
Hi,

If only the left eye is rendered, you need to change the **Stereo Rendering Mode** to Multi pass (see below).
[image].

Regarding the first issue, I remember fixing it by adding [this code ](https://www.youtube.com/watch?v=t_JQsA2z1CA). 
Can you make sure it reaches it on your project ? It's working fine on my side (tested on 2020.3 with the xr compatibility branch).

Best regards,
Benjamin Vallon

crud89, 2021-05-21:
Thanks for the reply! Interestingly switching from single pass to multi pass also fixed the movement issue (given the xr compatibility branch).

Bvallon-sl, 2021-05-21:
Hi,

good to hear ! Even if it was not expected at all.
When I have some time, I'll look at it, that's a strange behavior.

Best regards,
Benjamin Vallon

AndreiaValente, 2021-05-31:
> 
> 
> Hi,
> 
> The master branch has some compatibility issues with the XR Plugin management system. Can you try to use [this branch](https://github.com/stereolabs/zed-unity/tree/xr_compatibility_3.5) instead ? It should be working correctly.
> 
> Best regards,
> Benjamin Vallon

Hello Benjamin, 
I changed the branch and used 2020.3 and the drifting of the image was fixed but I still get the scene like it was rendering on a plane (example [HERE](https://youtu.be/U8b6IZOgq8M)). Any tips?

Bvallon-sl, 2021-06-02:
Hi,

The zed image will not fill the entire screen space because the field of view of the zed-m and your vr headset are not the same.

Trying to increase the size of the image will distort it, I don't recommend it.

Best regards,
Benjamin Vallon

closed 2022-04-27 by github-actions[bot]