stereolabs/zed-unity #183: ZED Mini Texture2D retrieval

ChiaraSapo, 2021-05-19:
Hi, I am using the ZED Mini Camera with Barracuda. I cannot understand how to retrieve the Texture2D of the camera from my code to modify the size of the image (I am trying to use the Tiny Yolo v2 neural network for object detection and I need to give a square image in input). Has anyone else tried this and could give me a suggestion on how to do that? Any help would be very appreciated. Thank you in advance!

Bvallon-sl, 2021-05-20:
Hi,

You can have access to both left and right images in the ZEDRenderingPlane.cs script which is attached to the left and right eyes.
The **TextureEye** attribute is automatically updated the new zed image. ([here](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Display/ZEDRenderingPlane.cs#L110)

I hope it helps.

Best regards,
Benjamin Vallon

sararom15, 2021-05-31:
Hi, 

I tried to retrieve the Texture2D of the ZEDcamera left in this way: 

        myZedMat = new ZEDMat();
        myZedMat.Create(1280, 720, ZEDMat.MAT_TYPE.MAT_8U_C4, ZEDMat.MEM.MEM_CPU);   

        // Retrieve ZEDCamera image from GPU for access in CPU
        Camera_Left.RetrieveImage(myZedMat, VIEW.LEFT_UNRECTIFIED, ZEDMat.MEM.MEM_CPU);

        // Create matptr to save pointer 
        IntPtr matptr = myZedMat.GetPtr(ZEDMat.MEM.MEM_CPU);
            
        int matwidth = myZedMat.GetWidth();
        int matheight = myZedMat.GetHeight();
        int len = matwidth * matheight * 4;
        byte[] frame_bytes = new byte[len];

        // Marshal copy pointer
        Marshal.Copy(matptr, frame_bytes, 0, len); 
        OurTexture = new Texture2D(1280, 720, TextureFormat.BGRA32, false);
        OurTexture.LoadRawTextureData(frame_bytes);

In addition, I've read from this link (https://github.com/stereolabs/zed-unity/issues/61#issuecomment-495416204) that the bytes are probably flipped, therefore I re-flipped the bytes in this way: 

  
            byte[] flippedbytes = new byte[len];
            int steplength = myZedMat.GetStepBytes(); 

            for(int i = 0; i < frame_bytes.Length; i += steplength)
            {
                Array.Copy(frame_bytes, i, flippedbytes, flippedbytes.Length - i - steplength, steplength);
            }
            
            OurTexture = new Texture2D(1280, 720, TextureFormat.BGRA32, false);
            OurTexture.LoadRawTextureData(flippedbytes);

Would it be correct? 
I needed the Texture2D from the ZED Mini because I'm using a Neural Network (Tiny Yolo v2) for Object Detection (which needs a Texture2D as input). 
However the results do not look very good, therefore I'm not sure what I did wrong. 
Could you give me any suggestion?

closed 2022-04-27 by github-actions[bot]