stereolabs/zed-unity #185: No depth occlusion not working with OpenGLCore

KalydosENSIIE, 2021-05-21:
Hello,
I want virtual elements to always appear in front real ones in my app. But when I turn the depth occlusion off on the ZEDManger, virtual elements seem to always be hidden instead. I think this problem has already been solved in this issue :
[](https://github.com/stereolabs/zed-unity/issues/75)
But it has been reverted in version 2.8.2.

I'm using :
Unity 2019.4.26
OpenGLCore
ZED_Unity_Plugin_v3.5.0.unitypackage

Bvallon-sl, 2021-05-21:
Hi,

Thanks for reaching us out.

It seems the fix has be reverted for some reason. You can re-add it manually ([this ](https://github.com/stereolabs/zed-unity/commit/27ccd584642ac1b53c351aff9f4ed2780ac7dc32)change). In the mean time, it'll add it in the plugin for the next release.

Thanks for the feedback.

Best regards,
Benjamin Vallon

KalydosENSIIE, 2021-05-21:
Alright, thanks :)