stereolabs/zed-unity #187: Unable to open camera

Kismetzc, 2021-06-05:
I followed the object detection tutorial [here](https://www.stereolabs.com/docs/unity/object-detection/).
However, when I build and run, it shows that unable to open camera.
I ran the body_tracking.py of zed-examples, and the camera can run normally. I don't know where the problem is.

* Ubuntu 18.04
* zed sdk 3.5
* zed -unity 3.5
* cuda 11.0

Kismetzc, 2021-06-05:
> 我按照[这里](https://www.stereolabs.com/docs/unity/object-detection/)的对象检测教程进行操作。
> 但是，当我构建并运行时，它显示无法打开相机。
> 我运行了zed-examples的body_tracking.py，摄像头可以正常运行。我不知道问题出在哪里。
> 
> * Ubuntu 18.04
> * Zed SDK 3.5
> * zed-unity 3.5
> * CUDA 11.0

Kismetzc, 2021-06-05:
Exception: Error: Unable to initialize Tracking module
ZEDManager.EnableTrackingThreaded () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:2305)
System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.ThreadHelper.ThreadStart () (at <695d1cc93cca45069c528c15c9fdd749>:0)
UnityEngine.<>c:<RegisterUECatcher>b__0_0(Object, UnhandledExceptionEventArgs) (at /home/bokken/buildslave/unity/build/Runtime/Export/Scripting/UnhandledExceptionHandler.bindings.cs:46)

Kismetzc, 2021-06-05:
Hi, @Bvallon-sl and @obraun-sl
I need your help!

Kismetzc, 2021-06-06:
> Exception: Error: Unable to initialize Tracking module
> ZEDManager.EnableTrackingThreaded () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:2305)
> System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <695d1cc93cca45069c528c15c9fdd749>:0)
> System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <695d1cc93cca45069c528c15c9fdd749>:0)
> System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <695d1cc93cca45069c528c15c9fdd749>:0)
> System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <695d1cc93cca45069c528c15c9fdd749>:0)
> System.Threading.ThreadHelper.ThreadStart () (at <695d1cc93cca45069c528c15c9fdd749>:0)
> UnityEngine.<>c:b__0_0(Object, UnhandledExceptionEventArgs) (at /home/bokken/buildslave/unity/build/Runtime/Export/Scripting/UnhandledExceptionHandler.bindings.cs:46)

Good news! Although there are some errors, but I run the demo successfully!
`Unable to open camera`may be the problem of data line contact

Kismetzc, 2021-06-07:
Hi,@Bvallon-sl
Does zed now support Maya?
I finally want to post-process the character model in Maya. I tried to use Unity's fbx export plugin on ubuntu, but an error was reported on ubuntu.

Bvallon-sl, 2021-06-07:
The Skeleton tracking sample support any Rigged 3D Model, it just needs to be rigged and its animation type set to "Humanoid" (see attachment).

You can modify the current model in Maya, or create a new one.

Best regards,
Benjamin Vallon

[image]

Kismetzc, 2021-06-09:
> The Skeleton tracking sample support any Rigged 3D Model, it just needs to be rigged and its animation type set to "Humanoid" (see attachment).
> 
> You can modify the current model in Maya, or create a new one.
> 
> Best regards,
> Benjamin Vallon
> 
> [image]

Hi, @Bvallon-sl 
Thanks for your instruction!
I create a model named fitness in MAYA. Then I import it into **zed.examples.skeletontracking.models**. And I follow your instruction to operate it. (see figure 1)

Next  I changed the parameters of **SkeletonTrackModule**'s **Avatar** to my model. (see figure 2) And I ran the demo. The error message is
```
MissingComponentException: There is no 'Animator' attached to the "fitness(Clone)" game object, but a script is trying to access it.
You probably need to add a Animator to the game object "fitness(Clone)". Or your script needs to check if the component is attached before using it.
```

I don't know what to do next and where to associate the model. Or is it necessary to make the model into a prefab, just like **ybot**?

I have never used unity before and I am looking forward to your help!

[image]

[image]

Bvallon-sl, 2021-06-09:
Hi,

First, once you imported your model, set its Animation Type as "Humanoid" as I showed you in my last message.
Then : 
 - Create a prefab of your model and add a Animator on it (same as Ybot as you said).
 - Replace the Avatar (like you did)
 - Press play

If you encounter errors as this point, make sure your model is rigged.

Best regards,
Benjamin Vallon

Kismetzc, 2021-06-09:
> 你好，
> 
> 首先，导入模型后，将其动画类型设置为“人形”，正如我在上一条消息中向您展示的那样。
> 然后 ：
> 
> * 创建模型的预制件并在其上添加 Animator（与您所说的 Ybot 相同）。
> * 替换头像（就像你一样）
> * 按播放
> 
> 如果您在这一点上遇到错误，请确保您的模型已被操纵。
> 
> 最好的问候，
> 本杰明·瓦隆

Thank you for your quick reply！It seems that I am missing the step of making a prefab and will try it right away.

Kismetzc, 2021-06-10:
> Hi,
>
> First, once you imported your model, set its Animation Type as "Humanoid" as I showed you in my last message.
> Then :
>
> * Create a prefab of your model and add a Animator on it (same as Ybot as [...]

I got the result. The actions recognized by the camera are inconsistent with the actions of the model. 
I'm thinking whether there is a problem with the animator or a problem with the skeleton.

[image]

Kismetzc, 2021-06-10:
This is the **avatar** and **Rig** of my model (see figure1 and figure2) . 
[image]
[image]

Bvallon-sl, 2021-06-10:
Hi,

I don't see anything wrong in your setup.
Can you first confirm that it's working as expected with the default avatar ?

Best regards,
Benjamin Vallon

Kismetzc, 2021-06-10:
> Hi,
> 
> I don't see anything wrong with your setup.
> Can you first confirm that it's working as expected with the default avatar?
> 
> Best regards,
> Benjamin Vallon

I replaced **ybot_avatar** with **fitness_avatar** in **SkeletonTrackerModule** and got the result mentioned above.
Our animation clip is different, but I saw the preview effect, this should not affect the result.

By the way, when I pressed the **play** button, it showed **Unable to open camera**. However, I tested the zed2 with the python version sample of body_tracking in  ZED SDK3.5, it worked well. I have tried restarting my computer and reconnecting to zed. Is it because the function of unity to connect to the camera is unstable on unbuntu？

Kismetzc, 2021-06-15:
> Hi,
> 
> I don't see anything wrong in your setup.
> Can you first confirm that it's working as expected with the default avatar ?
> 
> Best regards,
> Benjamin Vallon

Hi, @Bvallon-sl 
Good news!  I did it. The problem is that my model is bound with ADV instead of humanik.
I also want the character to stand on the floor and be able to see the whole body. Where do I need to modify？
[image]

Bvallon-sl, 2021-06-15:
Hi,

If the plane detection is correctly done, the Avatar should be standing on the floor.
Make sure the camera sees the floor enough.

Otherwise, if the camera will stay still, you can comment [this](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/SkeletonTracking/ZEDSkeletonTrackingViewer.cs#L211) line and manually set the position of the virtual camera in Unity.

Best,
Benjamin Vallon

Kismetzc, 2021-06-17:
> Hi,
>
> If the plane detection is correctly done, the Avatar should be standing on the floor.
> Make sure the camera sees the floor enough.
>
> Otherwise, if the camera will stay still, you can comment [...]

OK. It's an easy way.

I have another question now. (Don't mind I have a lot of questions hhh)

Can I bind the recognized action to the model and save the model with the action?

Bvallon-sl, 2021-06-17:
Hi,

One solution is to save your model + the animation as a FBX file. while you are playing it.
However, I don't know if it exists such a tool in Unity.

Best regards,
Benjamin Vallon

Kismetzc, 2021-07-01:
> Hi,
> 
> One solution is to save your model + the animation as a FBX file. while you are playing it.
> However, I don't know if it exists such a tool in Unity.
> 
> Best regards,
> Benjamin Vallon

Ask for your help again.
I don't know how to save the animation. Is it done through scripts?

Bvallon-sl, 2021-07-01:
Hi,

I think you need to look around some FBX writer to have a better understanding of the format and how/if it's usable in your case.

Best regards,
Benjamin Vallon

Kismetzc, 2021-07-01:
yeah

closed 2021-07-26 by Kismetzc

mraect, 2024-01-23:
> I followed the object detection tutorial [here](https://www.stereolabs.com/docs/unity/object-detection/). However, when I build and run, it shows that unable to open camera. I ran the body_tracking.py of zed-examples, and the camera can run normally. I don't know where the problem is. My hardware only has ZED2 and using

I did the same thing but instead followed ZED's "getting started" with Unity tutorial. Now I have black screen at the back of Unity example "Planetarium". 
I don't get any warning or error only black screen. When I pressed Play Button in the Game window, it says “Loading Camera…” and then “Loading Modules…” with black background, then I see the Scene Items but not the ZED 2i Camera output rendering only black screen at the back, also sensors works fine by the way.