stereolabs/zed-unity #188: Universal Render Pipeline problums

hagsac, 2021-06-06:
Thanks for your reply as always.
Is there any workaround?

[image]

When I import [Universal Render Pipeline  ], the following error occurs.

[image]

If I remove the above error, this error occurs.

[image]

And only Pink Screen is output on the screen.

hagsac, 2021-06-06:
what i am using.
[image]

In general plugin, everything worked normally.

Bvallon-sl, 2021-06-07:
Hi,
The last [commit ](https://github.com/stereolabs/zed-unity/commit/933717f0a821b0be94636c50ea95665d47b51225) should fix it.

Best regards,
Benjamin Vallon

hagsac, 2021-06-07:
When I import the ZED plugin, this message appears.
If you run the example after ignoring the messages, Pink Screen is displayed.

Are you sure you tested the modified plugin?
Currently I am spending a lot of time on various errors.

[image]

hagsac, 2021-06-07:
all fail

[image]

[image]

Bvallon-sl, 2021-06-07:
Hi,

You are right, I tested the wrong project on my side, I apologize.
This time, I fixed it and tested it, it is working correctly.