stereolabs/zed-unity #190: The following error occurred in object detection.

hagsac, 2021-06-12:
The following error occurred in object detection.
After setting up to detect only people in multi-class objects, a long time passed during testing and this phenomenon occurred.

[image]
[image]

Bvallon-sl, 2021-06-14:
Hi,

Can you try to set the SENSING MODE to **FILL** (see attachment) ? This way, the depth map will be complete and it won't return NaN values.

[image]

Best regards,
Benjamin Vallon

hagsac, 2021-06-15:
I'm always using the same [FILL] mode.
I recently made an application using 3D Detector.
As a result of the build, we recorded more than 3 hours afterward.
After a long time, the cube was created well, but the Tag ID was not printed (sometimes printed). It seems to be the same problem as above.
The error log could not be checked because it was executed as an exe.

hagsac, 2021-06-15:
Due to the size of the attached file, it is difficult to tell the details of the problem.
It is not possible with github mail.
Do you have any other email addresses?
hagsac@gmail.com

Bvallon-sl, 2021-06-15:
Hi,

You can send a message at support@stereolabs.com.

Best,
Benjamin Vallon

Bvallon-sl, 2021-06-15:
Hi again,

I'm not sure why this vector has Infinity value yet but I updated the code to avoid this error. You can pull the URP or Master branch.

Best regards,
Benjamin Vallon

closed 2022-04-27 by github-actions[bot]