stereolabs/zed-unity #191: I can't watch the controller(vive cosmos)in display, because the position of the camera and controller changes.

YuyaMurai, 2021-06-15:
I use a ZED mini and a head mounted display vive cosmos.
So, using the ZED's unity plugin, we are trying out shooting games and bat swinging games using controllers while watching 3D images on the head mounted display. However, when I start the game, camera_control object rotates and moves from the set camera position.
Moreover, I can't watch the controller(vive cosmos)in display, because the position of the camera and controller changes.
If I set a sphere or a block as a child object of camera_control, it will always be in the same relative position in the screen, but only the the controller(vive cosmos) will change its relative position when I start the game or point the camera in a different direction during the game.
How can I fix it?

Related to＃189

Bvallon-sl, 2021-06-15:
Hi,

If you disable the camera tracking (in the ZEDManager window), the camera rig won't move at all. However, any virtual objects in the scene will stay at the same relative position to the camera, even if you are moving.

Best regards,
Benjamin Vallon

YuyaMurai, 2021-06-16:
Thank you for nice advice!

After I started the game, the camera _eyes stopped working at all when I moved the camera. However, when I press the start button in the unity game, only the initial position and angle are different from the ones in the settings. This is the state of issue [#189](https://github.com/stereolabs/zed-unity/issues/189).
Do you know how to fix the weird values for the position and angle at the start of the game?

YuyaMurai, 2021-06-16:
Also, the IMU has been initialized and is in perfectly good condition.

Bvallon-sl, 2021-06-17:
Hi,

Can you be more specific about "the camera_eyes stopped working at all" please ?

I'm not sure I understand what you are looking for, the cameras positions and rotations are changing because the headset and the camera are tracked. If you move the ZED-mini camera, the virtual camera will also move, same for your VR headset.

You can disable all the tracking by commenting [this line](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L2497).

Best regards,
Benjamin Vallon

YuyaMurai, 2021-06-18:
Hi!

"the camera_eyes stopped working at all"  means that when I move the camera in the real world, the camera no longer moves in unity.　I'm sorry if I'm misleading you.

Anyway, The problem that is still unsolved is that when I start a game in Unity, even if I haven't moved the camera or HMD at all, "camera_eyes" will start at a position or angle that is different from the initial position or angle that you originally set.
[image]

[image]

>You can disable all the tracking by commenting this line.
I will check it later. 
Thank you.

YuyaMurai, 2021-06-22:
Thanks for good advice, Benjamin Vallon !
I was able to solve this problem.