stereolabs/zed-unity #192: Fresh install strange behaviour

SantoSimone, 2021-06-15:
Hi everyone, I am struggling running the examples of the plugin in my setup. Couldn't find anything about my problem so I'm posting here.

**Environment**
Windows 10 Home
Intel Core i9-11900K
64 GB RAM
NVIDIA GeForce RTX 3080 (Laptop)

Unity 2019.4.28f1 (also tested on 2020.3.11f1 and 2021.1.11f1)
ZED SDK 3.3.3 (minimum for CUDA 11.1, but also tested with 3.4.2 and 3.5.0)
ZED Unity Plugin 3.3.0 (also tested with 3.4.0 and 3.5.0, as above)
Oculus Integration 29.0

**What I am trying to do**
I would like to run the Planetarium example, just to make sure everything is working as it should, before switching to my project (already running on an older machine)

**What happens**
- On Unity 2019 game is not rendered on the oculus, giving a black screen, but in my editor it sees the passthrough video correctly.
- On Unity 2020 and 2021 Plugin version 3.3 and 3.4 give compilation error saying `the type or namespace name 'vr' does not exist in the namespace 'unityengine'` (not really concerned to solve this, see below)
- On Unity 2020 and 2021 I try with Plugin version 3.5 and the game starts correctly both on editor and oculus, but when i turn my head the screen does not follow me correctly, like the screen is actually turning faster than my head, falling out of sight at some point.

Am i missing some step in my setup? I went over it multiple times (with multiple configurations as you can see above), but didn't come to a solution.
Thanks to anyone for any help or pointer.

Bvallon-sl, 2021-06-15:
Hi,

Thanks for reaching us out.

If you plan to use the XR Management plugin, I recommend using the "xr_compatibility_3.5" branch available [here](https://github.com/stereolabs/zed-unity/tree/xr_compatibility_3.5).
It's on a separate branch for testing purpose but I'll probably merge it into master soon, it's not very clear.

For your other issue, you should try to set the Stereo rendering mode to "Multi-pass" (you can find more informations here : https://github.com/stereolabs/zed-unity/issues/182#issuecomment-845062442). 

Best regards,
Benjamin Vallon

SantoSimone, 2021-06-15:
That solved my problem, thanks!
Just a final comment: AFAIK, in new Unity versions the XR Management Plugin is the only way to use VR capabilities, so I would recommend adding this to the documentation.

Thanks again for the prompt solution!

closed 2021-06-15 by SantoSimone