stereolabs/zed-unity #193: Human skeleton recognition problem

yizes, 2021-06-22:
I want to use zed2 to judge some movements, such as waving and jumping. However, there seems to be no relevant example tutorial in unity Integration Tutorial. I wonder if there is any way to help me solve this problem faster.

Bvallon-sl, 2021-06-22:
Hi,

Thanks for reaching us out.

For the moment, we do not have gesture recognition features integrated to our skeleton tracking module.
You will need to use external ressources or develop your own detector based on data output by our Body tracking module.

Best regards,
Benjamin Vallon

nhruch, 2021-07-08:
Hey Benjamin,

Do you have any recommendations as to which external ressources we might be able to use for gesture recognition? Also, 
is gesture recognition on the development schedule?

Thanks for all your work and these examples. 

- Nick

Bvallon-sl, 2021-07-09:
Hi,

We do not plan to integrate gesture recognition in the ZED SDK for the moment.

I do not have external ressource to recommend but I think you might want to look into AI-based systems.
The leap motion cameras provides some gesture recognition, maybe you can look into it as well.

Best regards,
Benjamin Vallon

closed 2022-04-27 by github-actions[bot]