stereolabs/zed-unity #194: Bounding box problem

hagsac, 2021-06-22:
Hello. Thank you for your quick reply.
When a person is detected in a bird view, the bounding box is greatly enlarged.
I'm testing the floor at about 40 degrees, 7 to 8 meters high.
When a person moves to the bottom and edge of the camera view screen, the bounding box becomes significantly larger than the object size.  After setting the bounding box to Fixed, when tested, the results are judged normally in proportion to the sense of perspective. 
If this problem is not solved, 3D detection is meaningless.
It's a big problem with my project.

[image]
[image]

Bvallon-sl, 2021-06-25:
Hi,

The thing is the width and the length of the bounding box are the same, meaning that the top/bottom face are always a square, which can explain why the bounding box is so large.

What specific informations do you need for your project ?
Even if the box is large, you can have the 3D position of the person.

Best regards,
Benjamin Vallon

hagsac, 2021-06-25:
Good morning.
To help you understand, I will explain in comparison with unsetting the [transform scale] setting of the 3D Object Visualizer.
The [transform box scale] setting is a great feature.
So I want to use it.
However, due to the following large deviations, it is difficult to use.
In my opinion, this equates to losing the benefits of 3D sensing.

[image]

[image]

I'm making a safety system.
However, it is not easy to set in consideration of the increase in the bound box size depending on the location of the danger zone setting.

[image]

Bvallon-sl, 2021-06-25:
Hi,

Thanks for the detailed informations, I now have a better understanding of your situation.
I see two solutions : 
- Do not use the transform box scale, and only use boxes with fixed sized.
- Using the skeleton tracking model and create your own "minimal" bounding box with the joints position. 

 Best regards,
Benjamin Vallon

hagsac, 2021-09-03:
hi! 

Thanks for the sincere reply.
I've reviewed your answer.

Your solution first:
[image]

Your solution second:
Currently, the application I am making is processed in the SBC (Edge PC) unit.If you use a skeleton, it has been confirmed that it is very difficult to operate with low specifications.Shouldn't this be a problem that should be considered from the correction of camera angle and depth?

have a good day. 
chang in ho.

closed 2022-04-27 by github-actions[bot]