stereolabs/zed-unity #196: Image is upside down

KalydosENSIIE, 2021-06-25:
Hello,
I'm using zed-unity to make a mixed reality application on an oculus quest (used as a rift with oculus link), but the image from the camera is often displayed upside down on the HMD (not always). I'm very certain my camera is mounted right and I didn't have this problem at first. I made sure it was in the right orientation when starting the application too. I suspect it might be a disfunctionment of the sensors (which is quite problematic considering it happened on my two ZED Minis).

However, I don't need that function to automatically flip the image, so I would like to disable it. I see that there is an enum called FLIP_MODE in ZEDCommon.cs to define wether this option is enabled or not, but I don't see a way to change the FLIP_MODE in the ZEDManager.

Do you have a solution to my problem ?
Thank you for your help.

Bvallon-sl, 2021-06-28:
Hi,

Thanks for reaching us out.

You can modify the flip mode in the ZEDManager script [here](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L1074) and set it to OFF.

If it does not solve your issue, please send a mail at support@stereolabs.com .

Best regards,
Benjamin Vallon

KalydosENSIIE, 2021-06-28:
It works, thanks.
Maybe you should expose this in the editor.

closed 2021-06-28 by KalydosENSIIE

Bvallon-sl, 2021-06-28:
Ok,
Thanks for the feedback.

Best regards,
BV

SODAVISION-TECH, 2023-08-23:
Hi, may I know how can I set it to OFF in python language?