stereolabs/zed-unity #197: Camera stream blocking

PEPSoares, 2021-07-01:
Hi, 

I'm trying to use a ZED2  with unity and I'm having some problems.
I downloaded the latest zed 2 unity package.

Some demos work, like: 
Dark room , drone shooter greenscreen etc.... 

But the ones that I needed don't:
 Skeleton tracking and Object detection.

When I press play everything seems to work but only for a second or two. I can see the cam stream but then it blocks.
Sometimes unity fps counter keeps couting sometimes it doesn't but everytime I press play again to stop playing unity blocks and I have to kill the process.

On the console there is only 
Starting Object Detection. This may take a moment.
Object Detection module started in 6 seconds.

All the samples on the SDK work without any problem.
I'm using:
Win10 pro 64
Unity 2020.3.6f1
CUDA 11.1.0
ZED sdk 3.5.0 for cuda 11.1

And running on a 
NVIDIA GeForce RTX 3060 Laptop GPU

At first I got this error: 

[image]

Which stoped showing after I changed TdrDelay to 10s.
https://docs.microsoft.com/en-us/windows-hardware/drivers/display/tdr-registry-keys

Bvallon-sl, 2021-07-01:
Hi,

I've never encounter this error but, according to this forum post (https://forum.unity.com/threads/constant-crashes-failed-to-present-d3d11-swapchain-due-to-device-removed.987834/), it seems to happen to some people as well.

They are mentioning that deleting the Library folder might solve the issue. Did you try it ?

Best regards,
Benjamin Vallon

PEPSoares, 2021-07-01:
Thanks for the tip but unfortunately nothing changed.

Kismetzc, 2021-09-24:
The same question!