stereolabs/zed-unity #201: ChooseTrackedObjectMenu.cs(270,13): error CS0103: The name 'button' does not exist in the current context

noyco333, 2021-08-06:
Hello! I just got my ZED 2i to capture mixed reality videos for my Unity Oculus VR project.

Since I never used ZED before, I made an empty URP project in Unity 2021.1.16f, and imported the ZED Unity plugin for URP (by downloading the repository on URP branch because I couldn't find the unitypackage for it.)

Everything seemed to be working great until when I imported the Oculus integration (v29.0) from the asset store.

[image]

I also tried it on Unity version 2019.3.9f1 (which my project is made with) but it didn't work either.

Am I doing something wrong? Please tell me if more information is needed.

Thank you very much in advance.

noyco333, 2021-08-06:
I forgot to mention that when opening this project, a popup will appear saying that I should enter safe mode. This explains why many other scripts "could not be found".
[image]

noyco333, 2021-08-06:
I may be wrong but it probably boils down to this section of ChooseTrackedObjectMenu.cs

[image]

Bvallon-sl, 2021-08-16:
Hi,

Can you try to use directly the branch instead of the unity package ? You just need to download it and copy the folder in the Asset folder of your unity project. I may have made some fixes without recreating a new unitypackage.

Best regards,
Benjamin Vallon

noyco333, 2021-08-19:
Hello, I'm sorry for the late reply. I ran out of time and had to use an alternative system.
I will try this out later, thank you for the instructions!

closed 2022-04-27 by github-actions[bot]