stereolabs/zed-unity #202: Which two joint points world_joints_pos[18] and world_joints_pos[19] represent

Kismetzc, 2021-08-08:
I found in the process of reading the code [ZEDSkeletonTrackingViewer.cs](https://github.com/stereolabs/zed-unity/blob/6781d2ff9b859e60f8d42047819d1c256b77e5fa/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/SkeletonTracking/ZEDSkeletonTrackingViewer.cs#L241)

It seems that two joint points are calculated approximately, but I don’t know which two joint points are.

Kismetzc, 2021-08-08:
Hi, @obraun-sl 
I know in your previous answer that the ZED camera can recognize 18 joint points
[image]

Kismetzc, 2021-08-08:
And is the skeleton format in the picture is COCO-skeleton?

obraun-sl, 2021-08-09:
Yes that's the output format of the skeleton keypoints in the ZED SDK. 
In Unity , we arrange some points to match the rig of common avatars in Unity : 
- KP0 becomes the middle point between 16 and 17. (Head point). It replace the previous KP0 point given by the zed sdk which is reported in KP19.
- KP18 is the middle point between 8 and 11. 
- KP19 is the previous 0 point (Nose).

Kismetzc, 2021-08-09:
I converted it into a .bvh file based on the coco skeleton, so I just take k0-k17.
Another question is why world_ joints_ POS has only one precision  while in the `zed-sample BodyTrackging` it has 8 precision.

Kismetzc, 2021-08-13:
One question more, which point is the root？KP0(Head point) or KP19(Nose point). I guess it is KP19.

Kismetzc, 2021-08-13:
Is there a way to get the approximate coordinates of the Chest point?

Kismetzc, 2021-08-18:
Hello, @obraun-sl  and @Bvallon-sl，
I urgently need to know which is the root joint so that I can write a .bvh file.

Bvallon-sl, 2021-08-18:
Hi,

All the joint positions are in the camera coordinate frame, there is no root joint.

Best regards,
Benjamin Vallon

Kismetzc, 2021-08-18:
> Hi,
> 
> All the joint positions are in the camera coordinate frame, there is no root joint.
> 
> Best regards,
> Benjamin Vallon

Hi,
But aren’t these coordinates converted from camera coordinates to world coordinates?

Bvallon-sl, 2021-08-18:
Hi,

Yes exactly. I think we are not talking about the same thing here.
What do you mean exactly by "root joint" ?

Kismetzc, 2021-08-18:
> Hi,
> 
> Yes exactly. I think we are not talking about the same thing here.
> What do you mean exactly by "root joint" ?

A root joint needs to be defined in .bvh file. I guess the relationship between this root joint and the node is defined by myself

Bvallon-sl, 2021-08-18:
The output joints positions are in a global coordinate frame (not local, ie the position of a joint does not depend of the position of its parent), that is why there is not root joint here.

I think the bvh format requires local transforms (position + rotation) which is not what we output here. Therefore it needs a root joint.

You need to convert the joints positions (and rotation) to local space using the global position and the bones hierarchy.

Kismetzc, 2021-08-18:
> The output joints positions are in a global coordinate frame (not local, ie the position of a joint does not depend of the position of its parent), that is why there is not root joint here.
>
> I think [...]

Your idea is very correct. I should define a root joint, and then calculate the rotation angle and offset of other joints relative to it

Kismetzc, 2022-04-11:
Hello,@Bvallon-sl and @obraun-sl @

In this [Body Tracking Sample in Unity](https://www.stereolabs.com/docs/unity/body-tracking/), I want to use other 3D human pose estimation models, such as videopose3D, is there any way to do this?

I also see that the body tracking API is an extension of the object detection API does this mean that I need to use a custom object detection model first, as explained on [this page](https://www.stereolabs.com/docs/object-detection/custom-od/)

I am very interested and anxious about this question, which is very important for my next work. If you see it, please let me know the answer as soon as possible. Thank you！

Bvallon-sl, 2022-04-11:
Hi,

For the moment, the Custom Object Detection feature only computes 3D bounding boxes, not Skeleton data.

However, it seems that videopose3d already outputs 3D data, so you don't need to use the Custom Object module.
You can just ingest the images from the ZED in the model without using the Object detection module of the SDK.

Best regards,
Benjamin Vallon

Bvallon-sl, 2022-04-11:
Indeed, if you use an external AI model, you won't be able to use the Unity samples directly, you will need to adapt the interface between the display and the SK data.
Moreover, as you don't use the SDK at all for the skeleton detection, you won't have the skeleton tracking information (the detection ID for example). 
One solution would be to ingest the 2d bounding boxes to the Custom model to get the Tracking data in addition to your model (videoPose3d) for the 3D SK data .

It requires some work to make it work and it might not be real time at all, videopose3d seems relatively slow.

Kismetzc, 2022-04-11:
> 

OK, I see what you mean. I'll think about it again and look at some related APIs later. Your answer really helped me a lot, thank you very much，Benjamin Vallon!

Bvallon-sl, 2022-04-11:
Please next time create a new issue, it is easier to track.

Have a good day.

closed 2022-05-18 by github-actions[bot]