stereolabs/zed-unity #203: ZED/ZED Forward Lighting Shader

LeoReubelt, 2021-08-10:
I am trying to set up an ar app using the zed mini for passthrough video with an oculus quest 2. I created an empty test app that worked fine.  However, when I tried implement passthrough video in our existing ar app, the material is pink, and in the shader inspector, it says the following...

SRP Batcher  not compatible
Material property is found in another cbuffer than "UnityPerMaterial" (_MainTex_ST_)

When I run the app, the screen is pink.
My guess is that this has something to do with the Universal Render Pipeline, but I am new to unity, so that may not be the case.

Bvallon-sl, 2021-08-16:
Hi,

Indeed, if you are using the URP pipeline, you need to use the URP version of the unity plugin which is available on a separate branch (https://github.com/stereolabs/zed-unity/tree/URP_Compatibility_beta)

Best regards,
Benjamin Vallon

keli95566, 2021-09-25:
@LeoReubelt  Hi there! I am wondering if you are deploying your app directly to the Quest 2 or you are also just using SteamVR + Virtual Desktop? :)

closed 2022-04-27 by github-actions[bot]

adeckerroto, 2024-01-03:
Hey there, this link is now dead and I'm experiencing the same issue, is there a fix for this I should be looking for?

SLJLMacarit, 2024-01-04:
Hi, the branch is now merged.
To use our project with URP or HDRP, please follow the instructions here: https://www.stereolabs.com/docs/unity#step-1-installation.

You may need to update the ZED prefabs for URP/HDRP.

adeckerroto, 2024-01-04:
I have it fixed now, it should be noted that those instructions are a little off.
On 2022.3.8f1 I had to change "Upgrade all the Materials to URP" in order to get it to work.

What worked for me was selecting all materials (this is easiest by going to the ZED folder of the plugin, hitting the asset type button in the project view, which is the square circle triangle button, then selecting materials, and shift clicking all of them). Then I hit Edit -> Rendering -> Materials -> Convert Selected Build In Materials to URP. then all of the examples worked for me. the rest of the instructions worked fine