stereolabs/zed-unity #204: AR passthourgh frames slip away when rotating head on Oculus

senikovski, 2021-08-16:
Hi!

When I rotate my head with the HMD on, the picture frames in AR passthrough seem to rotate away from me way too fast. For example, when I rotate my head 45 degrees to the left the frames rotate way out of my FOV.

I use Oculus Rift with Zed M, SDK 3.5 with Unity 2021.1.13.

Thanks!

Bvallon-sl, 2021-08-17:
Hi,

Can you try to set the Stereo Mode to multi pass ? It seems to fix this issue (according to another user : https://github.com/stereolabs/zed-unity/issues/182#issuecomment-845748502).

Best regards,
Benjamin Vallon

senikovski, 2021-08-17:
Thank you so much! The issue was resolved by the latest XR compatibility branch merger. 

Another short question: Is it possible to change the surroundings around the passthrough frame, for example from black to the unity skybox?

Thanks!

Bvallon-sl, 2021-08-19:
Hi,

If you add theses lines :

```
        finalLeftEye.clearFlags = CameraClearFlags.Skybox;
        finalRightEye.clearFlags = CameraClearFlags.Skybox;
```

at [this ](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/MR/ZEDMixedRealityPlugin.cs#L385) location, it should do the trick.

Best regards,
Benjamin Vallon

senikovski, 2021-08-19:
Amazing thanks so much!

closed 2021-08-19 by senikovski