stereolabs/zed-unity #205: Flickering Images on Quest2 and Alienware

keli95566, 2021-09-02:
Hi there! Thank you very much for the great packages!
I am testing a ZED Mini + Oculus Quest2 setup. Zed Mini is connected to an Alienware with RTX 2070, windows 10. Running wirelessly via Virtual Desktop and Steam VR. Everything works fine until I close the Alienware laptop lid. The streaming becomes very slow with very low framerate and flickering images.
I already set the advanced power option of windows to "do nothing" after closing the lid. But I am still getting such errors. Could anyone please help me troubleshoot?

Running on Unity 2019.4 LTS, ZED SDK3.5, Cuda 10

Bvallon-sl, 2021-09-03:
Hi,

Thanks for reaching us out.

I think that when you close your laptop, the USB bandwidth are reduced (even with the power option mode) which result to bad performance.
I don't know if there is a solution to this outside of keeping the laptop open.

Best regards,
Benjamin Vallon

FrankSpalteholz, 2021-11-24:
I'm trying to get a Quest2 + Zed Mini running (PartyRoom Example) on Unity 2019.4.14f + Cuda 11.2 + ZED SDK 3.6. App is starting correctly via Oculus Link and in the Game View i can also confirm that cameras are running. But by looking through the Quest2 i have only black screen. I also made sure to switch to MultiPass (using the XR Managment though). So i'd super thankful to point out some initial steps to get this to work. Thank you very much!

closed 2022-04-27 by github-actions[bot]