stereolabs/zed-unity #207: Running on Oculus Quest2 via Android SDK

keli95566, 2021-09-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Since it is possible to run ZED MR application on Oculus Quest2 via Virtual Desktop and Steam VR, would it be possible then to run the MR application directly on the standalone Oculus Quest2 device?  We could stream the images via ZED SDK and use the streamed images as input for the standalone application. Could you please let me know if there exists already a small tweak to enable such functionalities? Thank you very much!!

Bvallon-sl, 2021-09-24:
Hi,

Thanks for reaching us out.
Unfortunately, the latency induced by the streaming might be too high for AR applications.
I don't think it's a viable solution. 

Best regards,
Benjamin Vallon

keli95566, 2021-09-24:
I understand. Is there any way we could overlay Oculus passthrough API together with ZED Mini? https://developer.oculus.com/experimental/passthrough-api/

keli95566, 2021-09-25:
Hi there! Actually, would it be possible just to get a minimal version of the ZED Mini SDK such that we stream ONLY the see-through images? For my applications, I only need the two RGB images, no embedded depth needed, no forward shading needed, therefore I don't think there will be too big latency if I use a 5G network. 

I could try to modify the current unity package and make a customer lighter streaming script. Since streaming images via TCP should be straightforward code, and perpahs there are already similar scripts available in the earlier version?  It would save me a lot of time figuring out video format, API, etc. Your help is greatly appreciated!!

keli95566, 2021-09-28:
> Hi,
> 
> Thanks for reaching us out. Unfortunately, the latency induced by the streaming might be too high for AR applications. I don't think it's a viable solution.
> 
> Best regards, Benjamin Vallon

Hi, thanks a lot for the quick response!

Would it be possible just to get a minimal version of the ZED Mini SDK such that we stream ONLY the see-through images? For my applications, I only need the two RGB images, no embedded depth needed, no forward shading needed, therefore I don't think there will be too big latency if I use a 5G network.

I could try to modify the current unity package and make a customer lighter streaming script. Since streaming images via TCP should be straightforward code, and perhaps there are already similar scripts available in the earlier version? It would save me a lot of time figuring out video format, API, etc. Your help is greatly appreciated!!

closed 2021-10-04 by keli95566