stereolabs/zed-unity #209: Minimum Version of the ZED Unity SDK

keli95566, 2021-10-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hi there!

I am wondering if it is possible to get a minimum version of the ZED SDK that can run on Android?
I do not need the depth information/ shadow etc at the moment, just the pass-through from the two cameras. I just need to have the images of the stereo camera streamed to a device such as Quest2. 

Thank you very much!!

guypystin, 2022-03-14:
> 

Hello, did you manage to install the ZED SDK for Android? I have the same problem now

guypystin, 2022-03-14:
> keli95566

Hello, did you manage to install the ZED SDK for Android? I have the same problem now

keli95566, 2022-03-14:
ZED SDK won't work with Android. However, we could build Gstreamer for Android and receive makes it work with the ZED Gstreamer pipeline.

guypystin, 2022-03-14:
> ZED SDK won't work with Android. However, we could build Gstreamer for Android and receive makes it work with the ZED Gstreamer pipeline.

I need to get the .SVO file so that I can process it on another CUDA machine. Is it possible to get .SVO with GStreamer? Thanks for answering

closed 2022-04-27 by github-actions[bot]