stereolabs/zed-unity #219: Absolute angle difference between zed's wolrd view and the real world

hagsac, 2021-11-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Thank you for your help as always.

The problem is difficult to understand, but please check it.
At my level, it was too difficult to confirm the experimental results.
But the result is certain.

There were no major problems when installing the Zed camera at a height of 1~2m from the floor.
However, at a height of more than 7m, I found a problem with the estimated floor angle, so I tested it at my level as much as possible.
Please find the root cause of the problem below.

There is a difference of about 4 degrees between the angle of the world view floor calculated by Zed and the angle of the real world floor. (When the height is 7m)
Only the x-axis was checked for the axis where the difference occurred, and the error on the other axes was not checked.

Because it is difficult to contact git by documenting all the procedures of the test,
Please understand what I have explained as a conclusion by summarizing it as briefly as possible.

In my opinion, it is judged that the root cause exists rather than the problem of the unity example.

In fact, verification may be required after installing the camera at a high height.
Please don't ever conclude the problem I've found by guesswork.
------------------------------

### Steps to Reproduce

1. Install zed camera with a height of over 7m
2. Run the unity exam
3. Create plane object (0,0,0)
4. Comparison of the actual floor with a slope of '0' and the plane object
5. Find the problem

### Expected Result

Expected to have problems......

### Actual Result

[image]

### ZED Camera model

ZED2i

### Environment

```shell
os : windows10
cpu : i7
gpu : 2070
```

### Anything else?

 ...

hagsac, 2021-11-18:
What I just learned
I have two cameras.
Check result with svo recorded by each camera
The correction values are reversed as shown below.

svo of zed 2i : x-axis rotation correction value = -0.04f
svo of zed 2 : x-axis rotation correction value = +0.04f

Bvallon-sl, 2021-11-30:
Hi,

Sorry for the late reply.

Could you send us the SVO at support@stereolabs.com , we'll take a loot at it.

Best regards,
Benjamin Vallon

closed 2022-04-27 by github-actions[bot]